mantidproject/mantid #27290: Marker lines cannot be hidden

AnthonyLim23, 2019-11-06:
We would like to hide the marker lines in workbench plots. The user may just want the label.

martyngigg, 2019-11-07:
This feels like a feature request. Is it really necessary for the release?

NickDraper, 2019-11-08:
Pushing this out of the release as it is entering it's final week and we won't have time for this.

NickDraper, 2020-02-04:
So I've looked into this ...

It turns out that matplotlib  does not support this, if you set the LineStyle to "None" or None, which should be acceptable, then Matplot lib gives you a Solid line instead.

Also the is_visible(bool) function does not do this.

Therefore I'm going to close this as not reasonably possible, I think using markers this way is abusing them to add text.

closed 2020-02-04 by NickDraper