mantidproject/mantid #27324: Consistency of Iqt Fit output workspaces

RichardWaiteSTFC, 2019-11-08:
### Expected behavior
I(Q,t) fit returns  two groups of workspaces: one to hold the results (one workspace per run) and another to hold the spectrum data, fit and difference (one for each spectra). Ideally there would be consistent naming convention of workspaces when they are fitted with sequentially and simultaneously and it would be clear which workspaces belonged to which spectra/run.

### Actual behavior
The output workspace names are very different for sequential and simultaneous fitting. The output workspaces from sequential fitting of spectra from one run (left) and two runs (right) are below
 
[image]

The workspaces corresponding to each spectra are labelled with a spectrum number but no run number, and the result workspace names are subtly different (aside from the MultiIqtFit prefix) there is an additional "Fit_seq" in the single run case.

The output workspaces from simultaneous fitting of spectra from one run (left) and two runs (right) are below
 
[image]

for the case of fitting spectra from multiple runs the workspaces are now labelled sequentially not by spectrum number.

### Steps to reproduce the behavior

(1) Run the following to load data

from mantid.simpleapi import *
Load(Filename='irs26174_graphite002_red.nxs', OutputWorkspace='irs26174_graphite002_red')
Load(Filename='irs26173_graphite002_res.nxs', OutputWorkspace='irs26173_graphite002_res')
TransformToIqt(SampleWorkspace='irs26174_graphite002_red', ResolutionWorkspace='irs26173_graphite002_res',
ParameterWorkspace='irs26174_graphite002_TransformToIqtParameters', OutputWorkspace='irs26174_graphite002_iqt')
Load(Filename='irs26176_graphite002_red.nxs', OutputWorkspace='irs26176_graphite002_red')
Load(Filename='irs26173_graphite002_res.nxs', OutputWorkspace='irs26173_graphite002_res')
TransformToIqt(SampleWorkspace='irs26176_graphite002_red', ResolutionWorkspace='irs26173_graphite002_res',
ParameterWorkspace='irs26176_graphite002_TransformToIqtParameters', OutputWorkspace='irs26176_graphite002_iqt')

(2) Open the Indirect data analysis interface > I(Q,t) Fit tab
(3) Load some spectra from the available workspaces and attempt to fit (e.g an exponeontial).

This is related to issue #27284 (which is currently being fixed temporarily by renaming the workspaces from the results workspaces in IndirectFitOutput.cpp. However a more general fix is required to sensibly name the output workpaces at the point the fit is done to address inconsistencies between simultaneous and sequential fitting.
### Platforms affected

RichardWaiteSTFC, 2019-11-08:
The naming seems to be done inside QENSFitSequential.cpp

AnthonyLim23, 2019-11-14:
@RichardWaiteSTFC does your fix mean we can close this issue?

RichardWaiteSTFC, 2019-11-14:
The fix does now name workspaces sensibly after they have been created, but I opened this issue in case at a later point we might want to change how the workspaces are named initially (i.e. when they are created I think in the two .cpp files i mention above).

SpencerHowells, 2019-11-14:
@RichardWaiteSTFC @AnthonyLim23 we need to have a thorough look at the naming

closed 2021-03-02 by stale[bot]