mantidproject/mantid #27346: Unable to load scripts containing bash prompts

ConorMFinn, 2019-11-11:
This is the second part of this: #27337 
### Expected behavior
Loads the script
### Actual behavior
Loading a script with the first 2 lines (commented out!!!!) as such:

```
#!/usr/bin/python
#-*- coding: utf-8 -*-
```
gives error: File "", line 0
SyntaxError: encoding declaration in Unicode string
### Steps to reproduce the behavior
Open Nightly Workbench

To reproduce try loading test_py23.py from [test_py23.zip](https://github.com/mantidproject/mantid/files/3831625/test_py23.zip)

It will fail to open! If you open it in a text editor and copy in the contents it will produce the aforementioned error on line 0.
### Platforms affected
All

martyngigg, 2019-11-11:
Sorry to dump this on you today.

If it helps I think this is a Python 2 only problem with `compile`/`exec` in the code editor.

For python 2 `lib2to3` can help us detect the encoding and we can encode it manually:

```python
from lib2to3.pgen2.tokenize import detect_encoding
from cStringIO import StringIO

lit = u"""\
    # -*- coding: utf-8 -*-
"""
exec lit # this fails

line_reader = StringIO(lit)   # this fails
exec lit.encode(detect_encoding(line_reader)[0])    # this passes
```

ConorMFinn, 2019-11-12:
> For python 2 `lib2to3` can help us detect the encoding and we can encode it manually:

Unfortunately, the detect_encoding method seems to be incompatible with any non ascii symbols, and so throws an error if a script containing one is loaded. The script provided in the issue uses a `£` which isn't compatible.

PR #27366 (2019-11-14): Allow Scripts Containing Encoding Declarations to be Loaded and Run in Workbench
**Description of work.**
Scripts containing an encoding declaration in the first two lines are now able to be run and loaded without errors or encoding artifacts.

When the script is first compiled a specific error is returned. This is caught, the encoding declaration is removed, and the script is recompiled to check for any other syntax errors. 

**To test:**

1. Load [script](https://github.com/mantidproject/mantid/files/3831625/test_py23.zip) into Mantid workbench.
2. Script should run correctly without errors  [... cut]
closed 2019-11-14 by NickDraper