mantidproject/mantid #27374: plt.show() not showing matplotlib figures in workbench

RichardWaiteSTFC, 2019-11-13:
### Expected behavior
The following script would show a figure window in workbench

    ws=Load(Filename="GEM40979.raw", SpectrumMin=431, SpectrumMax=750)
    ws=SmoothData(InputWorkspace=ws, NPoints=20)
    plt.plot(0.5*(ws.readX(0)[:-1]+ws.readX(0)[1:]), ws.readY(0), '-og')
    plt.show()

### Actual behavior
A figure is generated but it isn't shown. You either have to select show on the plot menu (as shown in screenshot) 
[image]

or use
    plt.gcf().show()

### Steps to reproduce the behavior
(1) Run the script above
(2) See if figure is visible

### Platforms affected
Workbench 4.1 and nightly as of 08-Nov-19

DanielMurphy22, 2019-11-13:
May be an IDaaS Issue!

DanielMurphy22, 2019-12-16:
Yes its still an issue on IDaaS, not an issue on Windows

NickDraper, 2020-01-31:
Also an issue in Direct CentOS machines according to Pascal

NickDraper, 2020-02-07:
This does happen on windows as well

PR #28699 (2020-05-19): Fix plt.show() Not Showing the Most Recent Figure
**Description of work.**
`plt.show()` was not working due to the global figure manager not being replaced early enough. This PR refactors a lot of the startup code out of `mainwindow.py` and into a `main.py` module. The code to switch the matplotlib global figure manager out for our own is now the first thing to be run on startup. 

**To test:**
1. Ensure the Matplotlib version you are using is >= 3.1.2. Windows should have this by default, Linux users can use a virtual python environment. 
 a. importing `matplotlib` and running `matplotlib.__v [... cut]
closed 2020-05-19 by NickDraper