mantidproject/mantid #27375: Create Multithreaded Macro to Set OpenMP Max Threads to the Value From the Config

ConorMFinn, 2019-11-13:
Due to some issues with how secondary threads on Linux don't seem to carry over their OpenMP configuration from the main thread, there should be a macro in the `MultiThreaded.h` file that allows you to set the properties of the current thread to what is in the config. 

Currently, the setting is done by the Framework Manager.

DanNixon, 2019-11-14:
Ideally, more investigation as to why this is a problem in the first place should be done before this is implemented.

DanNixon, 2019-11-14:
As per the OpenMP spec, *nthreads-var* (the ICV that is set by `omp_set_num_threads()` and friends) is scoped to the data environment, which looks to be scoped to thread (i.e. non-OMP thread, it would be inherited in that case).

As a result, calls to `omp_set_num_threads()` only affect the current and subsequent data environments within the current thread.
The `MaxCores` entry in the properties file likely never worked and that is validated by my own findings and reports from users.

The "best" solution would be to call `omp_set_num_threads()` on each createion of a new data environment that we control via the macros in `MultiThreaded.h` (i.e. on each new `parallel` region).

DanNixon, 2019-11-14:
Looking more finds this minimal example of why what we do doesn't work: https://stackoverflow.com/questions/56361293/overriding-omp-num-threads-from-code-for-real

ConorMFinn, 2019-11-19:
> The "best" solution would be to call `omp_set_num_threads()` on each createion of a new data environment that we control via the macros in `MultiThreaded.h` (i.e. on each new `parallel` region).

I just want to be clear on this. As part of the macros that create an environment (such as `PARALLEL_FOR_IF`) there should be a call(?) to the macro that sets the number of threads to the value in the config?

DanNixon, 2019-11-19:
> I just want to be clear on this. As part of the macros that create an environment (such as `PARALLEL_FOR_IF`) there should be a call(?) to the macro that sets the number of threads to the value in [...]

Yes, although not ideal this is what we'll have to do to make the macros behave as they are supposed to.

If there is a case where the old behavior is actually desired then the `PRAGMA_OMP` macro may be used.

PR #27517 (2019-12-06): Make OpenMP for loops use max threads value from config
**Description of work.**
Before each of the OpenMP parallel for macros, the max threads value is updated for the current scope to match the value in the config. 

This is a fix that can be generally applied to replace the solution from #27343.

**To test:**
1. Ensure that the value set in your local `mantid.user.properties`file for `MaxThreads` is lower than the number of cores on your machine.
2. Open the Indirect Data Reduction Interface
3. Load a workspace via the ISIS Energy Transfer tab (I've used 88888 as a run number). 
4. Switch t [... cut]
closed 2019-12-06 by martyngigg