mantidproject/mantid #27386: Merge In Updated Tube Calib Scripts

DavidFair, 2019-11-14:
**Original reporter:** ISIS/Richard

Richard has been recently working on a local copy of the tube calibration scripts (`scripts/calibration`) and would like the changes merged back in and tidied up.

Specifically, these changes work better where the resolution is bad. However, it has been changed with the horizontal tubes of SANS2D rather than the existing tubes (horizontal on MERLIN) in mind. So we need to check that it still produces sane results on instruments with horizontal tubes.

smk78, 2020-07-16:
NB: Richard says start with the Direct Beam iterator script first (https://github.com/mantidproject/mantid/issues/28986) as it will be easier!

gemmaguest, 2020-07-29:
See also #10892

rbauststfc, 2023-03-14:
Outstanding tasks as of 14/03/2023:

- [ ] Investigate options for providing parameters as an input file to make it easier to use and troubleshoot the large number of parameters the algorithm requires, as well as keep track of past parameters that were used. The algorithm will be run infrequently, so there needs to be an easy way to run it again using the correct parameter values. It should be possible to see both front and rear detector values together in a file for ease of use. If we were to try and make use of scripts produced by the history in Mantid then they would need to be easy for users to read.

- [ ] Save the cvalue printed output as a file, as this needs to be referred back to for deciding which tubes to mask. This must be output with the correct detector suffix to avoid overwriting (see task below).

- [ ] The calibration algorithm has to be run twice for SANS2D (once for rear detector and once for front) and because output workspace names are not unique they are overwritten on the second run. A lot of diagnostic workspaces are currently output, so we should identify which of these need to be retained and therefore shouldn't be overwritten.

- [ ] Tidy up `Background` parameter code - this parameter value is actually not used at all. In the code, the `FlatTopPeak` algorithm is used for the peak fitting and this algorithm uses a hard-coded value as the background. We need to confirm what this parameter is actually doing and it's likely we should make this use the value passed in via the Background parameter. The other place in the code that uses a background value is if a Gaussian function is used to fit the peak, in which case another hard-coded value is used. However, while the code for Gaussian fitting is still in the script, we’re not actually using it so I think we could remove it.

- [ ] We need to identify which hard-coded values can be retrieved from the IDF, which would help to make this more generic to include both Larmor and Zoom.
- [ ]  Both Larmor and Zoom need to be supported by the script.
- [ ]

closed 2023-09-06 by SilkeSchomann