mantidproject/mantid #27440: Unexpected exception when entering invalid range in Iqt fit

RichardWaiteSTFC, 2019-11-20:
### Expected behavior
When entering an invalid range the programme would notify the user and ask for another range without unexpected exception or crashing mantid.

### Actual behavior
In MantidPlot 4.2 and 4.1 entering an invalid range (e.g. 5-4) produces an error in the log

[image]
but also produces an unexpected exception and the Error reporting window
[image]

On workbench 4.2 entering an invalid range causes a runtime error and causes the window to be greyed out and hang indefinitely (had to be closed from task manager)

[image]

### Steps to reproduce the behavior
(1) Run following code to get workspaces to fit
```
from mantid.simpleapi import *
Load(Filename='irs26174_graphite002_red.nxs', OutputWorkspace='irs26174_graphite002_red')
Load(Filename='irs26173_graphite002_res.nxs', OutputWorkspace='irs26173_graphite002_res')
TransformToIqt(SampleWorkspace='irs26174_graphite002_red', ResolutionWorkspace='irs26173_graphite002_res',
ParameterWorkspace='irs26174_graphite002_TransformToIqtParameters', OutputWorkspace='irs26174_graphite002_iqt')
```
(2) Open the Data Analysis interface > I(Q,t) tab > Multiple Input and add workspace. Enter spectrum numbers with an invalid range (e.g. 5-4)

### Platforms affected
MantidPlot and Workbench

AnthonyLim23, 2020-02-24:
@Mantid-Matthew how important is this one?

Harrietbrown, 2020-03-02:
Fixed by #28158

PR #28158 (2020-03-02): Turn unhandled logic error into handled runtime_error
**Description of work.**

Attempting to use an improperly defined spectra range (e.g. 5-3)
causes appendValue in PropertyHelper.h to throw a logic error.  The
gui alerts the user to runtime errors, but fails over on logic errors.
We now catch the logic error and demote it to a runtime error.

**To test:**

Fixes #27440. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the cod [... cut]
closed 2020-03-02 by Harrietbrown