mantidproject/mantid #27449: Shortcut equivalence for Workbench, eg Ctrl+W

DanielMurphy22, 2019-11-20:
**Original reporter:** [MANTID]/[@PhilColebrooke ]

### Expected behavior
Shortcuts in MantidPlot should work in Workbench
### Actual behavior
Some such as Ctrl+W to close a script editor tab do not (yet Ctrl+N to open one does...)
### Steps to reproduce the behavior
Open Plot and Workbench and look at which shortcuts (Usually highlighted in the drop-down menu) work in Plot and not Workbench
[image]

### Platforms affected
Workbench all

refs #24115

mducle, 2019-11-21:
Also in the editor, the shortcuts are different - e.g. `Ctrl+/` in Workbench vs `Ctrl+M` in MantidPlot to comment lines - they should also be made consistent.

DanielMurphy22, 2019-11-22:
Here is a short list of some MantidPlot shortcuts. Ctrl+Shift+L (to clear all memory in Mantid) has been highlighted to me recently as a rather useful one.

closed 2021-03-01 by stale[bot]