mantidproject/mantid #27459: CalculateMonteCarloAbsorption for elastic scans

gvardany, 2019-11-21:
MonteCarlo absorption normally requires an input workspace in the x-units of wavelength.
However, for the elastic (and inelastic) fixed energy window scans, the x-unit is the scanning observable (e.g. temperature).
Currently the `CalculateMonteCarloAbsorption` transposes the workspace, if the x-axis is not in wavelength, assuming that the y-axis is in wavelength. It also supports the `_elastic` instrument definitions for (OS) IRIS.
To be able to handle also the ILL reduced data, the logic should be generalised as follows:
- If x unit is wavelength, go straight to calculation
- Else if it is in energy transfer or energy, convert to wavelength
- Else, take the Efixed from the sample logs, generate a dummy workspace in the unit of wavelength with a single bin corresponding to the Efixed (assuming elastic).

SpencerHowells, 2019-11-21:
@gvardany In my version in PP absorption I take Efixed from the IPF so it could be extended to read sample logs as well.

SpencerHowells, 2019-11-21:
@gvardany There is another complication when routines are used for Direct geometry in that they have a tendency to use Ei and not set efixed.

SpencerHowells, 2019-11-21:
@gvardany This is what I did earlier.
[CalculateElasticMCAbsorption.zip](https://github.com/mantidproject/mantid/files/3875766/CalculateElasticMCAbsorption.zip)

closed 2020-05-06 by martyngigg