mantidproject/mantid #27464: sliveviewer: non-orthogonal axes

NickDraper, 2019-11-21:


martyngigg, 2020-04-06:
I've have done some initial investigation into this. Matplotlib seems to provide facilities for drawing both the image and axes grid with some transform applied. The code to compute the required ["skew" matrix](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/NonOrthogonal.cpp#L108) already exists and can be lifted and placed into an appropriate location that is easily callable from Python.

Once the transform is known then the peaks coordinates can be transformed easily.

The line viewer needs to use `BinMD` with the correct basis vectors and the integrated signal will be given.

My estimate break down:
 - 3 days to get the transform code into the appropriate place with appropriate tests
 - 3 days to use the transform and ensure the data/axes are displayed correctly on the view with several data sets with different normalisations.
 - 2 days for peaks viewer + tests
 - 3 days for line viewer + tests

This brings my total estimate to 11 days. I'm tempted to round this to 15 to allow for things I've not thought of.

NickDraper, 2020-04-07:
So to give this some bounds it sounds like 11 days minimum, 15 most likely and something like 25 maximum?

martyngigg, 2020-04-07:
I would be happy with that.

DanielMurphy22, 2020-04-09:
Some MantidPlot issues worth checking aren't part of this new non-orthogonal sliceviewer:

Poorly cropped: #27962 and #22203
Doesn’t zoom on the area selected #27961
2D cuts Mantid vs Horace: [Image Comparison](https://www.mantidproject.org/Vates_Validation#Set_of_2D_cuts_for_non-orthogonal_lattice) #11368

martyngigg, 2020-05-26:
Estimate to add peaks overlay in non-orthogonal view.

PR #29125 (2020-08-12): Enable peaks overlay for the nonorthogonal slice viewer.
**Description of work.**

Adds support for peaks workspace overlays for non-orthogonal axes in the new slice viewer.

**To test:**

The following script creates a fake peak at a known HKL.

```python
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws = CreateMDWorkspace(Dimensions='3', Extents='-5,5,-5,5,-5,5',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto= [... cut]
closed 2020-08-12 by DavidFair