mantidproject/mantid #27475: Workbench Features to allow migration

NickDraper, 2019-11-21:
### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

AndreiSavici, 2020-05-05:
Issue #28650 is dealing with building Mantid on systems with Qt5 only

martyngigg, 2020-11-02:
@peterfpeterson The only two remaining items are ORNL items: #28490 & #28491. Can I transfer this to you to take forward?

AndreiSavici, 2020-11-02:
And Step Scan Analysis. That's on our TODO list

martyngigg, 2020-11-03:
Thanks. I've removed our ISIS team label to indicate we won't be considering it for work scheduling.

martyngigg, 2021-04-28:
It looks like the final two issues for this were killed by the stale bot. Should we close this and you can schedule your remaining items separately when they are a priority for you?

closed 2021-04-29 by martyngigg