mantidproject/mantid #27482: SANS Cannot use event data for transmission runs

DavidFair, 2019-11-22:
**Original reporter:** ISIS/Steve

### Expected behavior
The SANS GUI should be able to accept event mode data as a transmission run

### Actual behavior
SANSLoad falls over with the following stack trace:

```
Error in execution of algorithm SANSLoad:
SANSLoad: Cannot load event-mode data for transmission calculation. Attempted to load the file \\ndlzoom02\f\USERS\Schoofs_19_3\ZOOM00009947-add.nxs which is event-based as transmission data.
at line 137 in 'C:/MantidInstall-4.2.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSLoad.py'
caused by line 700 in 'C:\MantidInstall-4.2.0\scripts\SANS\sans\algorithm_detail\load_data.py'
caused by line 736 in 'C:\MantidInstall-4.2.0\scripts\SANS\sans\algorithm_detail\load_data.py'
caused by line 680 in 'C:\MantidInstall-4.2.0\scripts\SANS\sans\algorithm_detail\load_data.py'
caused by line 453 in 'C:\MantidInstall-4.2.0\scripts\SANS\sans\algorithm_detail\load_data.py'
caused by line 410 in 'C:\MantidInstall-4.2.0\scripts\SANS\sans\algorithm_detail\load_data.py'
```

Currently this can be worked around by processing the data (into its -add form) to trick the SANS GUI into using the right loaders

### Steps to reproduce the behavior
- Sum runs on the following Zoom data: 9947,9946,9925,9898
- Take the resulting add file and enter it into the following table fields:
- Sample Scatter / Transmission / Direct
- Set the user file to: TBC
- Hit process selected and observe the error

### Platforms affected
All

DavidFair, 2020-02-13:
Bump - this issue was reported again so moving to the top of the backlog

DavidFair, 2020-02-19:
Bump this issue was reported again by a different instrument scientists

However some additional input:
- Monitor 4 on SANS should already be in histogram mode 
- add files in event mode might not be valid to sum monitor counts

smk78, 2020-07-16:
Related to this is what happens (or, rather, doesn't happen) when event slicing. If you are slicing the SANS between t1 and t2, then really you should only be using the monitor spectra between t1 and t2 to calculate the transmission.

smk78, 2020-07-16:
Also see https://github.com/mantidproject/mantid/issues/13801

PR #31199 (2021-04-26): SANS Transmission - Support Added Event Data
*Requires  #31134 to be completed first*

**Description of work.**
Adds support for loading summed event data as the transmission (trans) run. We could already load raw trans data, but this was gated off.

**To test:**
- Open the ISIS SANS GUI
- Go to manage user directories
- Add system directories -> ISIS SANS -> ZOOM explicitly, it's not included by default
- Click user file and go to ISIS SANS -> ZOOM directory and pick any file available
- Go to sum runs
- Add 9946 twice, 9946 is raw event data
- Sum runs, you'll likely need to move the saved f [... cut]
closed 2021-04-26 by martyngigg