mantidproject/mantid #27500: ISIS Reflectometry GUI batch number increments even if previous batch no longer exists

giovannidisiena, 2019-11-26:
### Expected behavior
The batch number displayed should only increment if the previous batch exists, rather than every time a new batch is requested.

### Actual behavior
The batch number does not correspond to the actual number of batches, possibly caused by #26669 in response to #26534.

### Steps to reproduce the behavior
- Open ISIS Reflectometry
- Batch -> New
- Close `Batch 2`
- Batch -> New
- Observe it is now called `Batch 3`

### Platforms affected
All

gemmaguest, 2020-01-06:
It's worth asking scientists whether they care about this. I expect it will be low priority and might be a bit fiddly to fix so I'll ice-box this for now.

gemmaguest, 2020-07-20:
Closing because this is low priority and we are unlikely to get around to it. It was a developer suggestion rather than a request from users.

closed 2020-07-20 by gemmaguest