mantidproject/mantid #27504: Patch function on figure and axes deprecated

DanielMurphy22, 2019-11-26:
As far as I can tell, the way I'm plotting on matplotlib is old, but I can't use a newer way as this way uses projection = 'mantid' correctly
Worth working out the "correct way" and then allowing projection='mantid' to work with it!

Error read out: 

```
The figurePatch function was deprecated in version 2.1. Use `.Figure.patch` instead.
The axesPatch function was deprecated in version 2.1. Use Axes.patch instead.
The axesPatch function was deprecated in version 2.1. Use Axes.patch instead.
```

### Steps to reproduce the behavior 
Use this code to reproduce:
```
import matplotlib.pyplot as plt
from mantid import plots
from mantid.simpleapi import *

ws=Load('164199.nxs')

prop_cycle = plt.rcParams['axes.prop_cycle']
colors = prop_cycle.by_key()['color'] # 10 colors in default cycle

'''Change the following two parameters as you wish'''
custom_colors = ['#0000ffff', '#ff0000ff','#00ff00ff'] #I've chosen Blue, Red, Green
Number = 12 # How many Spectra to Plot

fig = plt.figure(figsize = (10,10))

ax1 = plt.subplot(211,projection='mantid')
for i in range(Number):
    ax1.plot(ws, specNum = i+1, color=colors[i%len(colors)])
ax1.set_title('Default')
ax1.legend()

ax2 = plt.subplot(212,projection='mantid')
for i in range(Number):
    ax2.plot(ws, specNum= i+1, color=custom_colors[i%len(custom_colors)])
ax2.set_title('Custom')
ax2.legend()

fig.suptitle('Line Plots: Color Cycle', fontsize='x-large')

fig.show()
```
### Platforms affected
Workbench 4.2

hsautessella, 2019-11-26:
Related issue? https://github.com/mantidproject/mantid/issues/27057

DanielMurphy22, 2019-11-26:
Closing as this seems to be the same as #27507

closed 2019-11-26 by DanielMurphy22