mantidproject/mantid #27534: Errorbars need Endcaps in Workbench

DanielMurphy22, 2019-12-03:
### Expected behavior
Plotting with Errorbars or adding errorbars to a plot, should include Endcaps. (As MantidPlot does)
### Actual behavior
By default the Endcap Size is 0, but can be increased in Figure Options
### Steps to reproduce the behavior
Plot Spectrum with errors, the end cap size is defaulted to 0. 

Endcapsize for all plots should be defaulted to a positive number (maybe 3, not sure if this works for all workspaces). Make sure to change this default for all methods of plotting with error bars (Normal 1D, Tiled, waterfall, from a script...)

### Platforms affected
Workbench 4.2 and before

hsautessella, 2019-12-03:
Just as a reference, this was a deliberate choice by Matplotlib to not plot caps by default. This was introduced in v2, so on RHEL (which uses v1.5) caps will be on by default. The ~~argument~~ discussion on the Matplotlib issue is quite entertaining https://github.com/matplotlib/matplotlib/issues/5047

PhilColebrooke, 2019-12-03:
Unrest in the matplotlib community 😨

DanielMurphy22, 2020-04-28:
File > Settings > Plots allows you to plot with errorbar caps at your own preference!!

closed 2020-04-28 by DanielMurphy22