mantidproject/mantid #27548: Autoscale turned off if you visit Figure Options

DannyHindson, 2019-12-06:
### Expected behavior

In workbench:

1. Load MAR11060.raw
2. Plot Spectra 1-10
3. open Figure Options and then close without changing any options by hitting OK
4. Run SumSpectra on the workspace with output workspace=input workspace

Plot should update to show single summed spectrum instead of 10 separate ones and should autoscale

### Actual behavior

Plot doesn't autoscale and summed curve is cropped (top portion not visible). Note that if you don't visit the Figure Options page in the middle of this process it does autoscale. Seems like the Figure Options page is turning off the autoscale feature. 

I did a bit of looking into this. I think it's caused by the Axes tab in the Figure Options dialog (mantid\qt\python\mantidqt\widgets\plotconfigdialog\axestabwidget\presenter.py), specifically this line in apply_properties:

ax.set_xlim(new_props.xlim)

This appears to turn off autoscaling. Perhaps apply_properties should return early if no properties have changed without reapplying all the limits (same as the curves tab does)?

### Steps to reproduce the behavior

See above

### Platforms affected

Was testing on Windows

samtygier-stfc, 2020-11-18:
This was fixed before in bbb6c024b777da8cb9f0132df2b2d7b272bac785 by storing self.current_view_props and checking for changes before calling set_xlim().

Then removed in 4c806e1c340fea021466ac021df19d296f498647 to fix #26676

So good fix needs to keep autoscale enabled if the user never changes the limits. And keep users limits if they have set them. I think checking for change is fragile, keeping track of an autoscale flag would be better.

Would it be good to add an autoscale tickbox to the axes diaglog? Limit entry fields could be disabled or even hidden if autoscale is enabled. Currently I think autoscale can only be re-enabled by click the 'home' button on the plot window.

DannyHindson, 2020-11-18:
@samtygier-stfc - just wanted to check whether that question was for me or Martyn? Probably best to check with Martyn on what solution to go for because it's more of a "core" issue even though I raised the issue originally. I probably picked this up in some pre-release testing back in Dec last year

samtygier-stfc, 2020-11-18:
On slack @martyngigg said a checkbox sounds reasonable. I'm working on this now.

commit 1b6bea3e (2020-11-19): Add axis autoscale checkbox to plot config dialog

Fixes #27548

Adds a checkbox to the plot config dialog and keeps it in sync with the plot autoscale options.

This avoids autoscaling getting accidentally disabled by the config box. Also makes it easy for the user to re-enable autoscaling
commit 71bedd7b (2020-11-25): Add axis autoscale checkbox to plot config dialog

Fixes #27548

Adds a checkbox to the plot config dialog and keeps it in sync with the plot autoscale options.

This avoids autoscaling getting accidentally disabled by the config box. Also makes it easy for the user to re-enable autoscaling
PR #29989 (2020-11-27): Add axis autoscale checkbox to plot config dialog
Fixes #27548

Adds a checkbox to the plot config dialog and keeps it in sync with the plot autoscale options.

This avoids autoscaling getting accidentally disabled by the config box. Also makes it easy for the user to re-enable autoscaling

**Description of work.**

This is to solve the issue of autoscaling being disabled by opening the plot config dialog. Now the autoscale setting for each axis is explicitly tracked, and and controllable with the check box. A previous approach to solving this by checking for changes to the limits ( bbb6c024b7 [... cut]
closed 2020-11-27 by martyngigg