mantidproject/mantid #27556: Different proton charge when using FilterEvents vs FilterByTime

gemmaguest, 2019-12-09:
The proton charge in a filtered workspace is different when using `FilterEvents` compared to `FilterByTime`, even though the events in the filtered workspaces are exactly the same. I would expect the proton charge to the the same given that the events are the same. I am not sure which result is correct, so this requires further investigation, but on the face of it the results from `FilterByTime` look more sensible so this may be a bug in `FilterEvents`.

### Steps to reproduce the behavior
This script demonstrates that the events are the same but the proton charge is different.
```
#
# This script does a comparison of the proton_charge in a filtered workspace
# using FilterByTime vs FilterEvents algorithms.
#

# The following line helps with future compatibility with Python 3
# print must now be used as a function, e.g print('Hello','World')
from __future__ import (absolute_import, division, print_function, unicode_literals)

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *

import matplotlib.pyplot as plt

import numpy as np

from datetime import *

# Set up a dummy workspace
ws = CreateSampleWorkspace("Event",BankPixelWidth=1)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:00:00", Value=1)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:10:00", Value=2)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:20:00", Value=3)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:30:00", Value=4)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:40:00", Value=5)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:50:00", Value=6)

# Extract the first 30 minutes  * 60 = 1800 seconds (roughly half of the data)
# First with FilterByTime, then with GenerateEventsFilter and FilterEvents
startTime=0
stopTime=1800
wsFiltered1 = FilterByTime(ws,StartTime=startTime,StopTime=stopTime)
GenerateEventsFilter(InputWorkspace=ws,StartTime=str(startTime),StopTime=str(stopTime),OutputWorkspace='split',InformationWorkspace='info')
FilterEvents(InputWorkspace=ws,SplitterWorkspace='split',InformationWorkspace='info',OutputWorkspaceBaseName='wsFilteredNew',FilterByPulseTime=True)
wsFiltered2=mtd['wsFilteredNew_0']

# Print the number of events and proton charge in the result.
# - we get the same number of events but the proton charge is different.
print("Filtering by %d - %d seconds" % (startTime, stopTime))
print("Number of events in input = %i" % ws.getNumberEvents())
print("  After FilterByTime      = %i" % wsFiltered1.getNumberEvents())
print("  After FilterEvents      = %i" % wsFiltered2.getNumberEvents())

print("Proton charge in input = %d" % ws.run().getProtonCharge(), ws.run().getProperty('proton_charge').value)
print("  After FilterByTime   = %d" % wsFiltered1.run().getProtonCharge(), wsFiltered1.run().getProperty('proton_charge').value)
print("  After FilterEvents   = %d" % wsFiltered2.run().getProtonCharge(), wsFiltered2.run().getProperty('proton_charge').value)

# Check that the actual events are identical in both results
# - we see that they are
def checkEventsMatch(ws1, ws2, specNo):
    spec1 = ws1.getSpectrum(specNo)
    spec2 = ws2.getSpectrum(specNo)
    ptimes1 = spec1.getPulseTimes()
    ptimes2 = spec2.getPulseTimes()
    tofs1 = spec1.getTofs()
    tofs2 = spec2.getTofs()
    if tofs1.size != tofs2.size or len(ptimes1) != len(ptimes2) or tofs1.size != len(ptimes1):
        raise RuntimeError("Different number of TOFs/ptimes")
        
    print("Checking %d events for spectrum %d: " % (tofs1.size, specNo))
    for eindex in range(tofs1.size):
        if ptimes1[eindex] != ptimes2[eindex]:
            raise RuntimeError("ptimes differ for index %d: %s  vs   %s" % (eindex, str(ptimes1[eindex]), str(ptimes2[eindex])))
        if tofs1[eindex] != tofs2[eindex]:
            raise RuntimeError("TOFs differ for index %d: %s  vs   %s" % (eindex, tofs1[eindex], tofs2[eindex]))
        if spec1.getWeights()[eindex] != spec2.getWeights()[eindex]:
            raise RuntimeError("Weights differ for index %d: %s  vs   %s" % (eindex, tofs1[eindex], tofs2[eindex]))
        if spec1.getWeightErrors()[eindex] != spec2.getWeightErrors()[eindex]:
            raise RuntimeError("Weights differ for index %d: %s  vs   %s" % (eindex, tofs1[eindex], tofs2[eindex]))
    print("  All event pulse times and TOFs matched for spectrum %d" % (specNo))
    print("  First event: %s, tof = %d" % (str(spec1.getPulseTimeMin()), tofs1[0]))
    eindex=tofs2.size - 1
    print("  Last event:  %s, tof = %d" % (str(ptimes1[eindex]), tofs1[eindex]))

checkEventsMatch(wsFiltered1, wsFiltered2, 0)
checkEventsMatch(wsFiltered1, wsFiltered2, 1)
```

Setting the `stopTime` to `1799`, `1800` and `1801` gives the following results. The `FilterByTime` results look more sensible to me but I am not 100% sure which is correct.

[image]

### Platforms affected
All

gemmaguest, 2019-12-09:
@DavidFair Diego was going to ask you about this - he thinks it may be related to the differences in the SANS GUI's compatibility mode

DavidFair, 2019-12-10:
I've checked the SANS workflow, were not affected by the above issue as we are using Filter By Time rather than Filter Events.

I think there has been a small miscommunication as he expects the data to produce the same values between compatibility mode on/off

closed 2021-03-02 by stale[bot]