mantidproject/mantid #27557: Tabs replaced by Spaces

DanielMurphy22, 2019-12-09:
**Original reporter** [James Lord]/[ISIS]

Workbench replaces tabs with four spaces whenever read through Python (ie if you print a string with a tab in it). It is NOT the case that pasting or opening in WB, displays tabs as spaces in the Script Editor. Only when Python reads it.

Run this in Workbench 

```
mystring= "hello	world" # a real tab
for i in mystring:
    print(ord(i))
print (repr(mystring))
```

Please see original [Mantid Forum Post](https://forum.mantidproject.org/t/workbench-replacing-tabs-with-spaces-in-string-literals/602)

DanielMurphy22, 2020-01-29:
Another example of this:
```
spaces="        " # some spaces
tabs="			" # some tabs (paste in)
print ("spaces", list(ord(x) for x in spaces))
print ("tabs", list(ord(x) for x in tabs))
```
Notice how 8 spaces are recognised as 8 spaces but 3 tabs are recognised as 12 spaces.

From the [Mantid Forum](https://forum.mantidproject.org/t/workbench-python-bugs-python-3-release/636)

NickDraper, 2020-02-24:
fixed by #27986

closed 2020-02-24 by NickDraper