mantidproject/mantid #27563: Fit Dialog Fixes

DanielMurphy22, 2019-12-10:
**Original reporter:** [James Lord]/[ISIS] on [Mantid Forum](https://forum.mantidproject.org/t/fit-dialog-in-workbench/614)

A few Issues with Fit Dialog functionality (access by executing the Fit algorithm)
- [ ] Tied parameters input to this Dialog don't seem to be obeyed (check in Algorithm History after the fit has run)
- [ ] Ticking "Keep Open" gives error message: `Property with given name already exists search object OUTPUTNORMALISEDCOVARIANCEMATRIX `. Maybe related to reusing an Algorithm object.
- [ ] Setting StartX and EndX on the Dialog, related to a certain workspace, then re-entering the Dialog (maybe you've already done a fit but didn't keep open. The suggestion is that reopening to the same assigned workspace should remember your last StartX and EndX inputs. I think this last point is particularly up for discussion as you may want those parameters reset upon re-entering the dialog.
- [ ] Check general funtionality of this Dialog for anything not included here.

### Platforms affected
Probably Both, spotted in workbench 4.2

DanielMurphy22, 2021-05-25:
@martyngigg should I label this up for Maintenance?

martyngigg, 2021-05-25:
@DanielMurphy22 Yes, please.

sf1919, 2021-11-25:
I've had a quick look at this one as I'm working on fitting functions for the Muon scientists and have some questions.

Where are the Create Output and Keep open boxes to tick?

Is the second issue still an issue? There have been some recent fixes to ties (#31600 and #31740) that may have resolved this?

DanielMurphy22, 2021-11-25:
[image]
If you execute the algorithm `Fit` then you get this dialog which is custom for this one algorithm. It's not been updated in a long time and as stated above there are a few issues. I can check to see if these tie fixes help for this dialog. Most of the reason to remove it is that users access fitting through the Fitpropertybrowser rather than this dialog.

closed 2022-06-10 by stale[bot]