mantidproject/mantid #27564: Poisson Nan values

DanielMurphy22, 2019-12-10:
**Original reporter:** [James Lord]/[ISIS] on [Mantid Forum](https://forum.mantidproject.org/t/fitting-minimizer-exception/613)

Worth reading the Forum post!

This issue can produce crashes when trying to fit.

### Expected behavior
If the minimizer calculates new trial parameters of NaN or Inf it should ideally try a more modest change, or explain clearly (by reporting Failed to Converge or similar) why my function is faulty.

### Actual behavior 
Poisson isn't handling negative values well.
Error message such as `Trying to set a NaN value (-nan(ind)) to parameter n`

### Steps to reproduce the behavior
Using a User Function and the **Poisson cost function** (and the Levenberg-MarquardtMD minimizer), to fit raw muon data. For example
Function='Name=UserFunction, Formula=1/(1/(n*(1+a*cos(x*8.16633892558+p))*exp(-x/tau)+b)+d), n=24133.0, a=0.2, p=-0.226184558207, b=0.01, d=0.00001, tau=2.19703'

Alternatively, James was also able to get his function in a python script: 
[load_and_fit_3.zip](https://github.com/mantidproject/mantid/files/3944975/load_and_fit_3.zip)

Running this script gives the error message. It can be run fine if you use Least Squares instead of Poisson.

James also mentioned: The diagnostics indicate that my function doesn’t ever get called with silly parameters, or ever return y<=0 which might make the Poisson cost calculator give NaN. But maybe the Poisson is subtly different such that the Jacobian matrix becomes singular and predicts a next good guess would be NaN?

### Platforms affected
Mantid 4.2, tried both in a script and with the Fit algorithm dialog

AnthonyLim23, 2019-12-11:
@alicerussell1 do you have any thoughts on this?

AnthonyLim23, 2021-03-01:
@StephenSmith25 do you have any thoughts on this? Is it worth reopening?

StephenSmith25, 2021-03-01:
Probably not worth re-opening as it likely will never be dealt with, given that fitting is to get a major rework at some point.

DanielMurphy22, 2021-03-01:
As comments have marked this as not stale, but @StephenSmith25 believes it is worth closing, that's what I'm doing.

closed 2021-03-01 by DanielMurphy22