mantidproject/mantid #27574: ApplyPaalmanPings crashes with workspace groups

gvardany, 2019-12-11:
In the following script `ApplyPaalmanPings` crashes with a segfault.

```
Load(Filename='corrections.nxs', OutputWorkspace='corrections_ass')

GroupWorkspaces(InputWorkspaces=['corrections_ass'], OutputWorkspace='corrections')

Load(Filename='fapi-fws.nxs', OutputWorkspace='fapi')

ApplyPaalmanPingsCorrection(SampleWorkspace='fapi', CorrectionsWorkspace='corrections', OutputWorkspace='wsfapicorr')
```

SpencerHowells, 2019-12-11:
@gvardany we have found that when sample & container are converted to wavelength the x-values are different in the 5th decimal place. Matthew is looking into it.

SpencerHowells, 2019-12-11:
@gvardany I've just run it from the gui with no problem. Corrections is a GroupWorkspace. Is the sample data a Group?
[image]

gvardany, 2019-12-11:
It is a reduced fixed energy scan, with several workspaces corresponding to different doppler energy kicks (0., 1.5, 3., 4.5, 6 etc). The corrections is a group however, as expected, and it is computed from a qens measurement. 
Apparently there are more than one problems: first, the sample workspace is a group, second it has temperature as x-axis. That's where it gets confused; it expects either wavelength or something convertible to it. I think we agreed once that it should not support groups, but somehow supplying a group still passes through the validation, which I don't understand yet how. 

I attach the files if you would like to reproduce (remove the .zip in the files).

[corrections.nxs.zip](https://github.com/mantidproject/mantid/files/3950665/corrections.nxs.zip)
[fapi-fws.nxs.zip](https://github.com/mantidproject/mantid/files/3950666/fapi-fws.nxs.zip)

SpencerHowells, 2019-12-11:
@gvardany We did agree not to support data groups. Have looked at the files & see your problem. Being strictly scientific the corrections will change with temperature via the density. As an approximation you could inhibit the checking if it is a fixed energy data set.
The problem we found occurred when using PG004 data for the first time - it's Ok for PG002! One solution is to rebin the can to sample after conversion to wavelength.

gvardany, 2019-12-11:
Good point about temperature dependence, however it is probably out of scope for the moment.

PR #27583 (2019-12-20): Avoid crashes in ApplyPaalmanPings
**Description of work.**

This enables the algorithm to run for fixed-window scans, where the input has temperature as x-axis.
Also catches a hard crash when supplying a workspace group.
For the moment it is agreed, that the algorithm will not accept groups.

**To test:**
1. Load the attached nexus files with this script:
```
Load(Filename='corrections.nxs', OutputWorkspace='corrections_ass')
GroupWorkspaces(InputWorkspaces=['corrections_ass'], OutputWorkspace='corrections')
Load(Filename='fapi-fws.nxs', OutputWorkspace='fapi')
```
2. This should work without [... cut]
closed 2019-12-20 by martyngigg