mantidproject/mantid #27593: Muon Analysis loading files

AnthonyLim23, 2019-12-13:
Open muon analysis
Set the search paths to include a dir with some files in, turn off search archive
Disconnect from the internet
Type the run number into the run box for your run
The path will be wrong - it may also be the wrong data

AnthonyLim23, 2019-12-13:
I have looked into this some more and have a test file. 

If you go to manage user directories and turn off the archive. and add the path to the test file. 

The set the GUI to MUSR and search for 62250, it will load some data
The data will be longitudinal, however the test file has been edited to be transverse field (works if loaded directly). 

It appears that something is not giving priority to the local paths.

AnthonyLim23, 2019-12-13:
it is the muon GUI that is doing this as the same thing does not happen when loading via Mantid's generic load.

AnthonyLim23, 2019-12-18:
I have had a look into this and the line causing the problem is https://github.com/mantidproject/mantid/blob/master/scripts/Muon/GUI/Common/utilities/muon_file_utils.py#L74

we will need to use file finder. We will also need to remove the use of file finder from increment step

giovannidisiena, 2020-02-27:
I have noticed that if you put a breakpoint in `handle_load_current_run` and hit the enter key then the current run loads, however the breakpoint is never hit. If you click the widget then the breakpoint is hit but there are issues with loading.

ConorMFinn, 2020-03-03:
Fixed by #28135

PR #28135 (2020-02-28): Fix File Paths After Loading in Muon Analysis
**Description of work.**
The muon GUI generates a path that is given to the LoadMuonNexus algorithm which allows it to check the archive and any selected data directories. That file path was what was shown in the GUI even if it was not an accurate way of showing where the loaded file is.

Now, the file path is pulled from the algorithm after the load, so is much more accurate than previously.

**To test:**
0. Message me for a modified version of `MUSR62250.nxs`
1. Place it somewhere not included in the list of user directories.
1. Open the Muon Ana [... cut]
closed 2020-03-03 by ConorMFinn