mantidproject/mantid #27594: Running algorithms in parallel crashes Mantid on RHEL7

mducle, 2019-12-13:
### Expected behavior

Running algorithm in parallel works.

### Actual behavior

Running two algorithms at the same time (e.g. using an GUI interface which runs some algorithms as part of its processing at the same time as running a script in the script window) crashes Mantid *on RHEL7 based distributions*.

### Steps to reproduce the behavior

Either run this script:

```
# The following line helps with future compatibility with Python 3
# print must now be used as a function, e.g print('Hello','World')
from __future__ import (absolute_import, division, print_function, unicode_literals)

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *

import matplotlib.pyplot as plt

import numpy as np

import threading

class loadThread(threading.Thread):
   def __init__(self, threadID, runno):
      threading.Thread.__init__(self)
      self.threadID = threadID
      self.runno = runno

   def run(self):
      print("Thread {}, loading: {}".format(self.threadID, self.runno))
      self.ws = Load(str(self.runno), OutputWorkspace=str(self.runno))
      
t1 = loadThread(1, 'MAR26837')
t2 = loadThread(2, 'MAR26838')

t1.start()
t2.start()
```

Or run a script which repeatedly runs some algorithm (e.g. `[Load('MAR{}'.format(rr), OutputWorkspace='MAR{}'.format(rr)) for rr in range(26800,26840)]`) and at the same time in two different script windows, or in a script window and at the same time use MSlice to load data / make cuts / slices. 

### Platforms affected

RHEL 7 based Linux distributions only (verified on CentOS 7.7 Core and RHEL 7.7 Maipo)

mducle, 2019-12-13:
Updated description after verifying that if neither of of the parallel algorithm are `Load` the crash does not happen. 

It seems to be something in the RH7 provided `libhdf5` (not thread-safe?). The backtrace from a crash is:

```
#0  0x00007f6c71b3c377 in raise () from /lib64/libc.so.6
#1  0x00007f6c71b3da68 in abort () from /lib64/libc.so.6
#2  0x00007f6c72bd5e26 in tcmalloc::Log(tcmalloc::LogMode, char const*, int, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem) () from /usr/lib64/libtcmalloc_minimal.so.4
#3  0x00007f6c72bd2764 in (anonymous namespace)::InvalidFree(void*) () from /usr/lib64/libtcmalloc_minimal.so.4
#4  0x00007f6c5cbaf442 in H5FL_blk_gc_list () from /lib/libhdf5.so.8
#5  0x00007f6c5cbaf0c5 in H5FL_blk_free () from /lib/libhdf5.so.8
#6  0x00007f6c5ce0457f in H5WB_unwrap () from /lib/libhdf5.so.8
#7  0x00007f6c5cc59c21 in H5O_cache_chk_load () from /lib/libhdf5.so.8
#8  0x00007f6c5cb3678c in H5C_load_entry () from /lib/libhdf5.so.8
#9  0x00007f6c5cb2dffe in H5C_protect () from /lib/libhdf5.so.8
#10 0x00007f6c5cb07701 in H5AC_protect () from /lib/libhdf5.so.8
#11 0x00007f6c5cc401e6 in H5O_protect () from /lib/libhdf5.so.8
#12 0x00007f6c5cc428fe in H5O_get_info () from /lib/libhdf5.so.8
#13 0x00007f6c5cbd3bc8 in H5G_loc_info_cb () from /lib/libhdf5.so.8
#14 0x00007f6c5cbe6e92 in H5G_traverse_real () from /lib/libhdf5.so.8
#15 0x00007f6c5cbe7ed6 in H5G_traverse () from /lib/libhdf5.so.8
#16 0x00007f6c5cbd3cd3 in H5G_loc_info () from /lib/libhdf5.so.8
#17 0x00007f6c5cc3bffa in H5Oget_info_by_name () from /lib/libhdf5.so.8
#18 0x00007f6c603823f8 in group_info1 () from /lib64/libNeXus.so.0
#19 0x00007f6c5cbd0482 in H5G_iterate_cb () from /lib/libhdf5.so.8
#20 0x00007f6c5cbd8b8d in H5G__node_iterate () from /lib/libhdf5.so.8
#21 0x00007f6c5cb0dcf2 in H5B_iterate_helper () from /lib/libhdf5.so.8
#22 0x00007f6c5cb0e0bd in H5B_iterate () from /lib/libhdf5.so.8
#23 0x00007f6c5cbe12ea in H5G__stab_iterate () from /lib/libhdf5.so.8
#24 0x00007f6c5cbdc4e7 in H5G__obj_iterate () from /lib/libhdf5.so.8
#25 0x00007f6c5cbd07b0 in H5G_iterate () from /lib/libhdf5.so.8
#26 0x00007f6c5cc2b8ea in H5Literate () from /lib/libhdf5.so.8
#27 0x00007f6c60383e87 in NX5getnextentry () from /lib64/libNeXus.so.0
#28 0x00007f6c60152722 in NeXus::File::getNextEntry() () from /lib64/libNeXusCPP.so.0
#29 0x00007f6c601564ca in NeXus::File::getEntries(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >&) () from /lib64/libNeXusCPP.so.0
#30 0x00007f6c60156937 in NeXus::File::getEntries() () from /lib64/libNeXusCPP.so.0
#31 0x00007f6c62d91716 in Mantid::Kernel::NexusDescriptor::walkFile(NeXus::File&, std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >&, int) ()
   from /opt/Mantid/bin/mantid/kernel/../../../lib/libMantidKernel.so
#32 0x00007f6c62d918cd in Mantid::Kernel::NexusDescriptor::walkFile(NeXus::File&, std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >&, int) ()
   from /opt/Mantid/bin/mantid/kernel/../../../lib/libMantidKernel.so
#33 0x00007f6c62d918cd in Mantid::Kernel::NexusDescriptor::walkFile(NeXus::File&, std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >&, int) ()
   from /opt/Mantid/bin/mantid/kernel/../../../lib/libMantidKernel.so
#34 0x00007f6c62d91c13 in Mantid::Kernel::NexusDescriptor::initialize(std::string const&) () from /opt/Mantid/bin/mantid/kernel/../../../lib/libMantidKernel.so
#35 0x00007f6c62d91f1d in Mantid::Kernel::NexusDescriptor::NexusDescriptor(std::string const&) () from /opt/Mantid/bin/mantid/kernel/../../../lib/libMantidKernel.so
#36 0x00007f6c57a6f3d0 in boost::shared_ptr<Mantid::API::IAlgorithm> const Mantid::API::(anonymous namespace)::searchForLoader<Mantid::Kernel::NexusDescriptor, Mantid::API::IFileLoader<Mantid::Kernel::NexusDescriptor> >(std::string const&, std::multimap<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > > const&, Mantid::Kernel::Logger&) () from /opt/Mantid/bin/mantid/api/../../../lib/libMantidAPI.so
#37 0x00007f6c57a6fd80 in Mantid::API::FileLoaderRegistryImpl::chooseLoader(std::string const&) const () from /opt/Mantid/bin/mantid/api/../../../lib/libMantidAPI.so
#38 0x00007f6c48910ac3 in Mantid::DataHandling::Load::getFileLoader(std::string const&) () from /opt/Mantid/plugins/libMantidDataHandling.so
#39 0x00007f6c489169ce in Mantid::DataHandling::Load::exec() () from /opt/Mantid/plugins/libMantidDataHandling.so
#40 0x00007f6c579e3134 in Mantid::API::Algorithm::exec(Mantid::Parallel::ExecutionMode) () from /opt/Mantid/bin/mantid/api/../../../lib/libMantidAPI.so
#41 0x00007f6c579ead53 in Mantid::API::Algorithm::executeInternal() () from /opt/Mantid/bin/mantid/api/../../../lib/libMantidAPI.so
#42 0x00007f6c57a09214 in Mantid::API::AlgorithmProxy::execute() () from /opt/Mantid/bin/mantid/api/../../../lib/libMantidAPI.so
```

martyngigg, 2019-12-16:
@mducle Thanks for the detailed error description. It does indeed point to an issue with HDF5. Has this just started happening or is it something that we have only just noticed?

mducle, 2019-12-16:
@martyngigg I think it's been around for a while. One of the system I tested it on only had Mantid 4.1 installed. I've only just noticed it when using MSlice whilst simultaneously running a reduction script in the background.

martyngigg, 2019-12-16:
@mducle Okay, that makes sense. I couldn't figure out what we would have changed to make this an issue so it certainly sounds like an issue with HDF5.

It looks like RHEL don't build the serial version of HDF5 with the [threadsafe flag](https://bugzilla.redhat.com/show_bug.cgi?id=1563616) where as [debian](https://sources.debian.org/src/hdf5/1.10.4+repack-10/debian/rules/#L97) do. We'll have to think about how to deal with this properly.

DanielMurphy22, 2019-12-16:
On Windows, this also seems to be a problem. Not had Mantid crash but stuck/hanging. When I forced it to close I got this exception:
[image]

DanielMurphy22, 2020-03-03:
This bug was reproduced by running reduce_mari_dos_2020.py while loading (I guess any file) but MAR27075.nxs

NickDraper, 2020-04-20:
This needs to be handled by better dependency management on Linux.

martyngigg, 2021-02-05:
The decision made by the [TSC](https://github.com/mantidproject/documents/blob/b4df39d3a8b4fbdb9db7ea3c6c8d7535b283e97f/Project-Management/TechnicalSteeringCommittee/meetings/2021/2021-01-12.md) has been to move to a conda-first means this will resolve itself.

I'll leave this open as an issue to think on any possible mitigation while that work is in progress as it could span a couple of releases to get all of the elements of it in place.

DanielMurphy22, 2021-09-27:
As @martyngigg mentioned above is not yet resolved so I'm leaving it open for reference

martyngigg, 2021-09-27:
This *should* be solved by the conda builds.

mducle, 2022-03-26:
Still an issue on nightly `20220324.2323` on IDAaaS.

martyngigg, 2022-07-22:
@mducle The new conda builds should hopefully have sorted this problem. The nightly on IDAaas is going to be switched to this soon so we should be able to verify it.

mducle, 2022-07-22:
@martyngigg So actually the new conda installation (of nightly only at present) is already live for new VMs. I just tested and the segfault does not happen with `6.4.20220721.1346` on IDAaaS but does occur for `6.4` (which is not conda-ized).

closed 2022-07-22 by mducle

martyngigg, 2022-07-22:
Ah that's good. I thought I'd just tried a new one and it wasn't there yet but I'm glad it is. Super!