mantidproject/mantid #27599: Figure Window Placement

DanielMurphy22, 2019-12-16:
**Original reporter:** [James Lord]/[ISIS]

In MantidPlot successively plotted spectra appear offset from eachother, whereas in Workbench they appear centrally ontop of eachother on the screen. 
While this is a matplolib default worth deciding if there is a fix we can make to aid workflow.

Should the Workbench figures appear offset? Or should there be an option for defining where on the screen plots appear? Currently this can be done in a script ([See Mantid help Forum Post](https://forum.mantidproject.org/t/matplotlib-figure-placement-on-screen/616/4)).

closed 2021-03-02 by stale[bot]

RichardHeenan, 2022-03-21:
I have the same question, 40 plots stacked directly on top of each other in middle of screen, very annoying as have to minimise or move all 40 of them.

RichardHeenan, 2022-03-21:
Hiding some plots in the widget helps in some circumstances.

fig.canvas.manager.window.move(400,100) does work, but has to be called before plt.show(), calling afterwards causes a hang, "Firgue not responding", and hangs Mantid
Now need to find out the valid range of coords for current canvas.