mantidproject/mantid #27610: Investigate requirements for Engineering Fitting tab

DannyHindson, 2019-12-16:
### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

ConorMFinn, 2020-01-23:
Workflow:
1. Load a focused data files and have it plotted in the interface (I'm thinking of using Muon analysis 2 for inspiration)
1. Loaded data files should have the option of being plotted together or being able to choose between them. (My current thinking is a list of run numbers that can be ticked or all selected)
a. Data should be presented as points without a line.
2. The peaks to be fitted should be selected from the plot. 
3. Fit using `PseudoVoigt` conveluted with `BackToBackExponential`
4. Overlay the fit over the data for checking. 
5. Data should be output as a single file for the fitting parameters for each run and the logs associated with this run. Some 3D representation of the data may be necessary, such as those provided by `.hdf` files.

ConorMFinn, 2020-01-23:
From the above workflow, required elements are:
- The ability to load a range of files. 
- A way to distinguish between runs and select the ones to plot. 
- A plotting space to select peaks and show the fitting. 
- A method for saving the output of the fitting of each run and each set of associated logs to a single file.

ConorMFinn, 2020-02-18:
## Fitting Workflow
### Peak Selection
1. Select by clicking points on the plot. 
2. Select using a text field to enter peak centres in d-spacing. This field should be updated with the actual values for the peak centre, once the fit is complete. 
### Data Layouts
1. Data is split into distinct peaks. The user should be able to select those peaks and the ranges to perform the fit. 
2. Two or more peaks are too close to one another to fit separately. They need to be fitted within the same range.

DannyHindson, 2020-06-30:
Requirements now captured from Saurabh

closed 2020-06-30 by DannyHindson