mantidproject/mantid #27623: Grey out HKL range if using PredictFractionalPeaks with modulation vector

RichardWaiteSTFC, 2019-12-18:
### Expected behavior
When using the algorithm with PredictFractionalPeaks with modulation vector(s) rather than HKL offsets, the HKL min/max are ignored. If this is the desired behaviour it is confusing that the user can still view and change the min/max values.

[image]

### Actual behavior

Grey out the HKL min/max if MaxOrder > 0 and at least one ModVector provided.

### Steps to reproduce the behavior
(1) Load in some peaks e.g.
```
LoadNexusProcessed(Filename='TOPAZ_3007.peaks.nxs', OutputWorkspace='peaks')
LoadIsawUB(InputWorkspace='peaks', Filename='TOPAZ_3007.mat')
IndexPeaks(PeaksWorkspace='peaks')
```
(2) Execute PredictFractionalPeaks in GUI

### Platforms affected

martyngigg, 2019-12-18:
I've unassigned myself as I think this should just go into the pool of work for SX and then be picked up and prioritised at the next sprint planning meeting.

DannyHindson, 2020-02-13:
Complete

closed 2020-02-13 by DannyHindson