mantidproject/mantid #27703: Print button Workbench Plotting

DanielMurphy22, 2020-01-13:
**Original reporter:** [Adrian Hillier]

### Expected behavior
Print button on a plot should bring up a dialog to allow printing (presumably exporting to an image file)
### Actual behavior
Button seems to do nothing on Windows
### Steps to reproduce the behavior
Plot a simple plot in Workbench
### Platforms affected
Mantid Workbench all versions including the Nightly
Seems to be a windows issue (could also be a Mac issue!)

PhilColebrooke, 2020-01-21:
[image]
This appears to no longer be an issue on Windows ¯\\\_(ツ)_/¯

DanielMurphy22, 2020-01-28:
The print button works on Windows builds but not on the Py3 Nightly Release. Could be a packaging issue.

martyngigg, 2020-01-29:
Starting workbench from a package build in a command line reports:

```QPrintDialog: Cannot be used on non-native printers``

but not for a developer build. I wonder if we are missing a plugin from the package.

martyngigg, 2020-01-29:
It does indeed look like we are not shipping the `printsupport` plugins with the Windows package. @DanielMurphy22 Could you confirm whether the print button works on macOS from a package?

DanielMurphy22, 2020-01-29:
Just checked, the print button does not work on MacOS packages.

martyngigg, 2020-01-29:
@DanielMurphy22 Thanks. I've created a pull request to fix Windows packaging. We can open a separate one for macOS.

martyngigg, 2020-01-29:
It looks like the macOS packaging suffers similar issues with plugin.

martyngigg, 2020-01-29:
Confirmed that this is **not** a problem on Linux packages as we don't package Qt ourselves.

NickDraper, 2020-02-21:
now have merged PR's for both windows and mac

closed 2020-02-21 by NickDraper