mantidproject/mantid #27739: Instrument View for OFFSPEC incorrect

DanielMurphy22, 2020-01-17:
**Original reporter:** [Jos Cooper]/[ISIS] [Mantid Help Forum](https://forum.mantidproject.org/t/instrument-view-in-workbench/628)

### Expected behavior
Instrument Viewer shows the detectors correctly and all views (Full 3D, Cylindrical and Spherical) work correctly.
### Actual behavior
For OFFSPEC files, it seems that Full 3D works, but spherical and cylindrical seem odd (parts are greyed out) and the SidebySide view crashes MantidPlot. (When I ran in WB I think I say a zero length vector error which may relate to parts of the detector with no defined geometry?)
### Steps to reproduce the behavior
As per the Forum post, Load OFFSPEC00050000.nxs and try viewing the WLSF detector.

There are really three parts to this issue, for OFFSPEC files:
- [ ] Sidebyside view does not show / crashes Mantid
- [ ] The Spherical and Cylindrical views don't seem correct.
- [ ] When you view the WLSF detector on the Instrument Tab and then change the view in the Render tab (say between spherical x and cylindrical x, for example) the WLSF detector doesn't seem to be selected any more, it seems like the LarmorDetectorSection

This issue may just need fixes relating to the IDF. If any further testing needs done, then please contact @DanielMurphy22 

### Platforms affected
Workbench 4.2

gemmaguest, 2020-01-29:
I've added an initial estimate of 1 day just to do some investigation into this. If it turns out to be a significant amount of work I'll need to work out where it sits in our priorities.

giovannidisiena, 2020-02-03:
I have just observed similar `sidebyside` behaviour with INTER as well.

giovannidisiena, 2020-02-03:
After initial investigation it appears that the `DetectorBench` component and `idlist` are causing the crash in `sidebyside`. Any changes made to the IDF will not be applied if you load the `.nxs` file however changes can be observed if you load the `.raw`.

giovannidisiena, 2020-02-07:
Still not made much progress with this yet. If you remove `DetectorBench` from the IDF and add the relevant ID lists it is possible to click `sidebyside` without crash, however you should see `Assembly [component] not flat` error. This does not seem to make much sense because, as far as I can see, the IDF should be working.

This is a crude test script I have been using to attempt to diagnose what's going on.
[offspec_idf_test.zip](https://github.com/mantidproject/mantid/files/4171810/offspec_idf_test.zip)

gemmaguest, 2020-08-06:
I would like to at least find out what's wrong here because sorting out the IDFs for reflectometry is a priority, although depending on the problem this specific case might not be. Putting estimate 1 for an initial investigation

PR #29797 (2020-10-26): Fix Instrument View Bug for Rendering Flat Panels
**Description of work.**

NR Scientists identified a bug in the rendering of the OFFSPEC instrument geometry which caused a crash in Mantid when trying to render the side-by-side view. 

**To test:**

- Launch Mantid and Load any OFFSPEC data file. We have a few files in our testing data.
- Launch the instrument view and select the side-by-side view in the render tab.

Mantid should not crash and a sensible side-by-side view should be rendered.

Fixes #27739. 

---

#### Reviewer ####

Please comment on the following ([full description](http:// [... cut]
closed 2020-10-26 by gvardany