mantidproject/mantid #27744: Python 3: Fix remaining tests under Python 3.7/3.8/matplotlib 3.1

martyngigg, 2020-01-20:
#27742 attempted to turn on Python 3 support by default but shows that tests are currently broken running under Python 3.7 or Python 3.8 and also matplotlib 3.1.

This issues is to fix these tests.

DavidFair, 2020-01-21:
To Do:
- [ ] **python.WorkflowAlgorithms.BayesQuasiTest.BayesQuasiTest (Failed)**
- [ ] python.WorkflowAlgorithms.BayesStretchTest.BayesStretchTest (Failed)
- [ ] MantidQtWidgetsCommonTestQt5_QtJSONUtilsTest (Failed)
- [ ] MantidQtWidgetsMplCppTestQt5_ColorsTest (Failed)

I've bolded the one I'm currently working on

DavidFair, 2020-01-21:
@martyngigg `MantidQtWidgetsMplCppTestQt5` is related to how we use sip, [possibly a fix?](https://stackoverflow.com/questions/51693993/pyqt5-error-pycapsule-getpointer-called-with-incorrect-name) but unsure how it fits in to how were configuring

martyngigg, 2020-01-22:
@DavidFair Thanks for looking at that. I thought I had taken the sip configuration into account in https://github.com/mantidproject/mantid/blob/27744_Fixup_py3_test_failures/qt/widgets/common/src/Python/Sip.cpp#L27. We have to support both situations where sip is bundled inside PyQt and not. Are we missing something there?

martyngigg, 2020-01-22:
We now have 3 system tests that are failing:
- [ ] SANSILLReductionTest.ILL_D33_LTOF_Test
- [ ] SANSILLReductionTest.ILL_D33_Test
- [ ] SANSILLReductionTest.ILL_D33_VTOF_Test

The results match quite closely but are not identical. It is not clear why.

martyngigg, 2020-01-22:
@gvardany alerted me to changes he has made in #27734 and merging these changes with this branch causes the above SANS ILL tests to pass. This means no further work should be required now.

PR #27761 (2020-01-23): Fix Windows Python 3 test failures
**Description of work.**
Fixes multiple Python 3 issues on Windows, these primarily come from small incompatible changes as our packages don't quite match Ubuntu's

**To test:**
- Ensure the unit test suite and system tests complete

Fixes #27744 . 

*This does not require release notes* because it is part of an epic moving to Python 3

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code co [... cut]
closed 2020-01-23 by gemmaguest