mantidproject/mantid #27767: IBEX alarams

AnthonyLim23, 2020-01-24:
IBEX only records the value of a sample log when the value changes. This means that if a detector (e.g. temperature) breaks, then it is not reflected in the nxs log values. We should add a mechanism to remove these bad values from the sample log. 

I have spoken briefly to Freddie and he has said that they will add (might already exist) a vector to signify if the detector is working or not.

NickDraper, 2020-04-21:
Ideas for storage in the workspace:
1. Load the data in LoadEventNexus
1. If there are any invalid values for a log create an int based filter log (like period), but named %log_name%_invalid_filter.  Do NOT create the filter if everything is valid.
1. In PropertyManager::filterByProperty (line 174)  Add a check to look for a matching _invalid_values log for each log
1. if found add that to the existing filters

This should (hopefully):
1. By default filter out invalid values
1. Not clutter up the logs with lots of invalid_filter additional logs unless they are needed.
1. Be a simple way to either use these to highlight errors in the GUI, or even just leave them in the list.
1. the existing approaches for accessing unfiltered data for a log should still work.
1. Will only take additional memory space up for logs with invalid values  (I was a bit worried about the SNS proton charge logs).

martyngigg, 2020-04-22:
I think this sounds sensible. Agreed that we don't want to add extra data for each log if it is not necessary. A couple of questions:

> Idea for storage in the workspace:
> 
>     1. Load the data in LoadEventNexus, warning to the log if invalid values are found

Is this only applicable for Event data?

>     2. If there are any invalid values for a log create an int based filter log (like period), but named %log_name%_invalid_filter.  Do NOT create the filter if everything is valid.

I'd like to avoid external things relying too much on the exact form of the name. Can we add methods like `hasInvalidValuesFilter` and `getInvalidValuesFilter(logName)` to avoid the raw string being used in most places? 

>     3. In PropertyManager::filterByProperty (line 174)  Add a check to look for a matching _invalid_values log for each log
>     4. if found add that to the existing filters

NickDraper, 2020-04-22:
@martyngigg Good point about the event data, although I missed on the explanation the change wi actually be in LoadNexusLogs, so it should cover all nexus based logs.

And yes I'll minimize the use of the bare string.  hasInvalidValuesFilter(logName) and the get method will have to be on the LogManager class as that is the only class that knows the full list of logs.

martyngigg, 2020-04-22:
> @martyngigg Good point about the event data, although I missed on the explanation the change wi actually be in LoadNexusLogs, so it should cover all nexus based logs.
>
> And yes I'll minimize the use [...]

Sounds good :+1:

PR #28564 (2020-05-21): Framework support for ISIS nexus log alarms

Adds additional logs alongside the affected log if invalid data is found.  Otherwise, nothing else changes.
If invalid data is found then the logs are automatically filtered just as for periods and running status.

**Report to:** @AnthonyLim23

**To test:**
1. Open workbench
1. Load ENGINX00228061_log_alarm_data.nxs from the unit test data directory
1. the log should get messages like this
   ```
   Load started
   Some "cryo_temp1" log data is invalid!
   Some "cryo_temp2" log data is invalid!
   Load successful, Duration 0.66 
   ```
1. right clic [... cut]
closed 2020-05-21 by gemmaguest