mantidproject/mantid #27823: Permissions Denied error message

DanielMurphy22, 2020-01-29:
Please Post on the [relevant Forum post](https://forum.mantidproject.org/t/loadmask-algorithm-execution-error/635) when this is resolved.

### Expected behavior
If a file which needs to be accessed by Mantid, has its permissions denied, then the Error Message displayed to the user should say something along the lines of "Permissions Denied for File:..."

### Actual behavior
This was found for a script which ran the `LoadMask` algorithm and said that the algorithm failed:
The error is: Error in execution of algorithm LoadMask: SAXParse Exception: no element found in line 1 column 0. Unable to parse File: in “/user/local/mprogs/InstrumentFiles/merlin/mask_Dec2019.xml”
### Steps to reproduce the behavior
Not sure of a good test case as the related file has had its permissions corrected, but essentially run `LoadEmptyInstrument` for the relevant instrument and then `LoadMask` on a mask file with its permissions denied.

### Platforms affected
All

DanielMurphy22, 2020-02-11:
Pas reported this error message which is misleading!!! 
```
"/archive/ndxwish/Instrument/data/cycle_19_4/WISH00047174.s01" is not a valid RAW file.
Error in execution of algorithm LoadRaw:
Incorrect file type encountered.
```

martyngigg, 2020-10-06:
@DanielMurphy22 Can I just clarify here that the issue is that if files are not readable by the user then we get these confusing error messages?

DanielMurphy22, 2020-10-06:
@martyngigg yes, and the solution is to improve the error message to say that! I had to remind myself of this problem, and my  longer message on the [linked forum post](https://forum.mantidproject.org/t/loadmask-algorithm-execution-error/635/3) is the clearest explanation.

closed 2020-11-05 by martyngigg