mantidproject/mantid #27846: Python Warnings are Mantid Errors

DanielMurphy22, 2020-01-31:
Currently, it is possible to produce many Python errors in a row, which make the new Error Notification service rather annoying! #27568 
### Steps to reproduce the behavior
Load OFFSPEC50000.nxs, produce a colorfill plot and then scroll out of the plot! You will see a warning
"Attempted to set non-positive left xlim on a log-scaled axis.
Invalid limit will be ignored." in Py3 and
 "Attempted to set non-positive xlimits for log-scale axis; invalid limits will be ignored." in Py2.
This scrolling allows the errors to make many many notifications in no time!

### INFO
When I searched for these error message in our files I found them in these 2 locations:
C:\mantid\external\src\ThirdParty\lib\python3.8\Lib\site-packages\matplotlib\axes\_base.py on line 3252 -> cbook._warn_external('')
C:\mantid\external\src\ThirdParty\lib\python2.7\Lib\site-packages\matplotlib\axes\_base.py on line 3156 -> warnings.warn('')
[Both related to self.get_xscale()]

Essentially Mantid is treating all Python warnings as Errors. We currently don't individually capture Python warnings as they are just tracked at StandardErrors, but capturing the errors ourselves may be the solution here. 

The suggested outcome is that Python warnings are logged as Mantid warnings and so the Error Notification service isn't called in cases like this!

[This link](https://docs.python.org/3/library/warnings.html) (especially catch_warnings at the bottom) may of of some use!
### Platforms affected 
Workbench all, but only a noticeable problem since the Notification Service

NickDraper, 2020-02-04:
I've just tried and failed to reproduce this with my dev build and the current nightly

DanielMurphy22, 2020-02-07:
This specific case for creating many Python warnings is no longer possible (at least not as easily) since #27741 was merged. To recreate, now need to run LoadEventNexus with NumberofBins set to 1, which is no longer the default. If this becomes annoying again in the future, it should be reopened, but for now this case has been resolved so I'm closing.

closed 2020-02-07 by DanielMurphy22