mantidproject/mantid #27881: Instrument View scrolling between files

DanielMurphy22, 2020-02-06:
**Original reporter:** James Lord
### Expected behavior
Data sets should be reloaded cleanly to allow the pointer to Instrument View to survive.
### Actual behavior
Reloading files with the Muon Analysis GUI in Workbench can crash Mantid when Instrument View is open for theses files!
### Steps to reproduce the behavior
Open Workbench
Open Muon Analysis GUI
Type in 62260 as the run number
Show Instrument on the "EMU62260_raw_data MA" workspace
Enter the Pick tab and click on some detectors
go back to the Muon Analysis GUI and use the <> buttons to move to the next run, and then back to 62260. Crash.
### Platforms affected
Workbench all

giovannidisiena, 2020-02-11:
I have been having a look at this and think the error is due to the use of `boost::weak_ptr` which loses its last reference and is automatically deleted. There is a catch block that throws a nullptr warning https://github.com/mantidproject/mantid/blob/2f7908d5c762530dd875e2833e616748fa706898/qt/widgets/instrumentview/src/InstrumentActor.cpp#L813-L814 however `m_maskWorkspace` is not actually set to be a `nullptr`. Adding this single line above the warning thrown in the catch block prevents the observed crash however it also means that `getMaskWorkspace()` is not guaranteed to return a valid pointer, as suggested by https://github.com/mantidproject/mantid/blob/2f7908d5c762530dd875e2833e616748fa706898/qt/widgets/instrumentview/src/InstrumentActor.cpp#L284-L292. @AnthonyLim23 I have very limited knowledge of this part of the codebase but am happy to implement a fix if possible. Since you have more experience with muon analysis maybe you can figure out if I'm on the right track?

AnthonyLim23, 2020-02-11:
There are two ways we can fix this bug.

1. Not replace the workspace in muon analysis
2. Fix the pointer in instrument view

The second option is the proper fix. I think @martyngigg would be able to say if your suggestion is along the right track. It sounds reasonable to me.

martyngigg, 2020-02-11:
@giovannidisiena I think you are on the right lines but that's not quite the place to nullify the mask workspace pointer. The instrument widget listens for replace notifications and should be able to replace itself properly. I thought the whole `InstrumentActor` was recreated when a workspace was replaced but maybe this is not happening:

https://github.com/mantidproject/mantid/blob/2f7908d5c762530dd875e2833e616748fa706898/qt/widgets/instrumentview/src/InstrumentWidget.cpp#L1334 suggests that if the old and new pointers don't match then the whole actor is replaced which should reset the mask workspace pointer. Do you know which way the code goes?

giovannidisiena, 2020-02-11:
@martyngigg As far as I can tell the exception is being thrown at line `232` and caught by `811` in `InstrumentActor.cpp` (as above) because `sharedWorkspace = m_workspace.lock()` is null. From what I've observed the call to `m_instrumentActor->getWorkspace()` falls over but then nothing is caught by:
https://github.com/mantidproject/mantid/blob/2f7908d5c762530dd875e2833e616748fa706898/qt/widgets/instrumentview/src/InstrumentWidget.cpp#L1339-L1340
hence `resetInstrument(resetGeometry)` is not ever called. Does that seem right to you?

martyngigg, 2020-02-11:
@giovannidisiena Thanks for the info. I'm not sure I 100% follow. Could we have a look when you are back in the office?

giovannidisiena, 2020-02-11:
@martyngigg of course! I'll be back tomorrow 🙂

PR #27922 (2020-02-18): Reload data sets in Muon Analysis without crash
**Description of work.**
This PR allows data sets to be reloaded within the Muon Analysis GUI (Workbench) without crashing Mantid while the Instrument View is open.

**To test:**
- Open Workbench
- Open Muon Analysis GUI
- Type in 62260 as the run number
- Show Instrument on the "EMU62260_raw_data MA" workspace
- Enter the Pick tab and click on some detectors
- Go back to the Muon Analysis GUI and use the <> buttons to move to the next run, and then back to 62260
- Should **not** crash

Fixes #27881 

---

#### Reviewer ####

Please comment on th [... cut]
closed 2020-02-18 by martyngigg