mantidproject/mantid #27883: Unable to edit script while a plot is open

gvardany, 2020-02-06:
It has been reported that in workbench, when a plot is open, it is impossible to click and edit the script in the script window. This has been observed on Ubuntu.

NickDraper, 2020-03-03:
So this does not appear to be an issue on windows.

after executing this script
``` python
from __future__ import division
import numpy as np
import matplotlib.pyplot as plt
from mantid import plots
from mantid.simpleapi import CreateWorkspace

# Create a workspace that has a Gaussian peak
x = np.arange(20)
y0 = 10.+50*np.exp(-(x-10)**2/20)
err=np.sqrt(y0)
y = 10.+50*np.exp(-(x-10)**2/20)
y += err*np.random.normal(size=len(err))
err = np.sqrt(y)
w = CreateWorkspace(DataX=x, DataY=y, DataE=err, NSpec=1, UnitX='DeltaE')

# Plot - note that the projection='mantid' keyword is passed to all axes
fig, ax = plt.subplots(subplot_kw={'projection':'mantid'})
ax.errorbar(w,'rs') # plot the workspace with errorbars, using red squares
ax.plot(x,y0,'k-', label='Initial guess') # plot the initial guess with black line
ax.legend() # show the legend
fig.show()
```

you can still edit the script with the plot open.

NickDraper, 2020-03-03:
But it can be reproduced on idaas.  This might be associated with the always on top changes made by @alicerussell1

PR #28222 (2020-03-10): Workbench: Set QTool flags on figure window and add option to disable
**Description of work.**

Attempts a different method at keeping plots on top of the Workbench window to avoid issues with window managers being confused when we tried to force the figure window to the top of the stack. Some like Xfce did not ever relinquish focus back to the main window.

Ideally this would go on to master at this point in the dev cycle but the problem exist on IDAaaS and is not so esoteric as first thought so we have little choice but to fix it.

**To test:**

A new setting "Plots on Top" has been introduc [... cut]
closed 2020-03-10 by NickDraper