mantidproject/mantid #27893: Update Beam Centre Modle Rmin/max based on HAB/LAB

DavidFair, 2020-02-07:
Different instruments want a different `r_min` and `r_max` value depending on the instrument and bank selection within `beam_centre_model`

I've asked Steve to find out from each instrument what their optimals would be, so we can include them for all

DavidFair, 2020-02-19:
We need to re-check this after the scientists have explored how various sizes interact following #27966

smk78, 2020-07-16:
On review, the SANS Group would like to specify the r_min, r_max, and iteration_step in the User File. So also see https://github.com/mantidproject/mantid/pull/28513

smk78, 2021-02-23:
Given that the radius limits have been propagated in the new TOML user files
```
[detector]
  radius_limit = {min = 0.035, max = 0.419}
```
the linked issues here need review.

closed 2021-08-29 by stale[bot]