mantidproject/mantid #27915: Table errors and plotting

AnthonyLim23, 2020-02-12:
Reported by James

If you create a table workspace using the script:
```
# The following line helps with future compatibility with Python 3
# print must now be used as a function, e.g print('Hello','World')
from __future__ import (absolute_import, division, print_function, unicode_literals)

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *

import matplotlib.pyplot as plt

import numpy as np

table = CreateEmptyTableWorkspace()

table.addColumn(type="int", name = "Run")
table.addColumn(type="double", name = "B")
table.addColumn(type="double", name = "T")
table.addColumn(type="double", name = "f0.A")
table.addColumn(type="double", name = "f0.AError")
table.addColumn(type="double", name = "f0.B")
table.addColumn(type="double", name = "f0.BError")
table.addColumn(type="double", name = "f0.C")
table.addColumn(type="double", name = "f0.CError")

for run in range(10):
    row = {"Run": run, "B": float(run), "T":float(run)+300.,
           "f0.A": float(run)*0.1, "f0.AError": float(run)*0.01,
           "f0.B": float(run)*0.2, "f0.BError": float(run)*0.02,
           "f0.C": float(run)*0.3, "f0.CError": float(run)*0.03}
    table.addRow(row)
           
```
Then set the run column to X values and the remaining columns to Y and Y error as appropriate. The try and create a line with error plot for `f0.B` Mantid will say there are no associated errors. I think this is because we have different number of y and y error columns.

DannyHindson, 2020-03-02:
I was about to pick this issue up but I can't reproduce this? I do get a plot with some error bars on it when I follow the instructions:
[image]

DannyHindson, 2020-03-02:
Need to have columns B and T also set to Y (but these cols don't have any corresponding error cols)

DannyHindson, 2020-03-04:
I've produced a fix for this. The issue was also present in 4.2 so will mention that on the PR so we can double check we want to put this in to the release at this stage

closed 2020-03-09 by NickDraper