mantidproject/mantid #27920: Read in value_strings of logs produced in ORNL NeXus files

peterfpeterson, 2020-02-12:
**Original reporter:** ORNL/Krikham

On POWGEN, `BL11A:SE:SelectSampleTemp/value_string` is a string version and the `value` field is an enumeration. Read in the string as a second `TimeSeriesProperty` with the name `BL11A:SE:SelectSampleTemp_string`.

wdzhou, 2020-09-03:
**2020.09.02** Triaged to ORNL RSE diffraction backlog [issue #68](https://code.ornl.gov/sns-hfir-scse/diffraction/diffraction-backlog/-/issues/68)

wdzhou, 2020-09-09:
From whom reported this issue.
```
Thanks for following up on this.  It’s been a while, but I think remember the request.  My main concern was getting the information from that PV in our run-summary files, and that’s already in there now.  (If I remember correctly, Pete helped me figure that out.)  The PV is also available in Mantid, though it shows just as a number, instead of the text.  But I think what we have now is good and meets our needs.
```
The issue is thus closed.

closed 2020-09-09 by wdzhou