mantidproject/mantid #27943: Mantid crashes after any error

gvardany, 2020-02-17:
Normally, when there is a runtime error raised by an algorithm, Mantid should display it, but not crash.
It seems that this is broken now:
```
ws1 = CreateSampleWorkspace()
ws2 = CreateSampleWorkspace()
ws2.setDistribution(True)
ws1 - ws2
```
This could be related to the addition of the feature for system notifications.

### Platforms affected
Observed on macOS, could be platform dependent.

martyngigg, 2020-02-18:
This appears to be an issue with desktop notifications on MantidPlot and Qt4 on some versions of macOS. I suggest we only enable desktop notifications for Qt >= 5. I think that means just wrapping https://github.com/mantidproject/mantid/pull/27568/files#diff-fdabe3c97b39edfabe753ac2200f226cR242-R247 in an `ifdef` for Qt >= 5 so that the notficaition code never even runs on MantidPlot.

gvardany, 2020-02-18:
I agree. Indeed the crash happens deeply inside the qt libraries. What I found interesting is that it crashes on the second passage over `showMessage`: when there is an invalid property for an algorithm, first the property throws and error (this is notified fine), and then the algorithm throws (it segfaults there). I was wondering if there might be some sort of race condition that can be explained by two almost immediate calls?

gvardany, 2020-02-18:
One should also edit the https://github.com/mantidproject/mantid/pull/27568/files#diff-8d4f76a3d29eaecd3f1713bf5a195892 to not to build the new class at all for Qt4.

martyngigg, 2020-02-18:
> I agree. Indeed the crash happens deeply inside the qt libraries. What I found interesting is that it crashes on the second passage over `showMessage`: when there is an invalid property for an algorithm, first the property throws an error (this is notified fine), and then the algorithm throws (it segfaults there). I was wondering if there might be some sort of race condition that can be explained by two almost immediate calls?

Interesting. Perhaps it is some time of problem writing to the display from a separate thread that is fixed in Qt5.

> One should also edit the https://github.com/mantidproject/mantid/pull/27568/files#diff-8d4f76a3d29eaecd3f1713bf5a195892 to not to build the new class at all for Qt4.

Good point. Thanks.

PR #27956 (2020-02-19): Disable desktop notifications for Qt 4
**Description of work.**

They currently cause MantidPlot to crash occasionally on macOS. They are intended as an enhancement for Workbench so disable them in MantidPlot.

**To test:**

Try the script in the attached issue in both MantidPlot and Workbench on macOS and ensure the desktop notifications only appear for Workbench.

Fixes #27943, #27858 .

*This does not require release notes* because **it was introduced in this release.**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/Reviewing [... cut]
closed 2020-02-19 by NickDraper