mantidproject/mantid #27948: SANS Not Applying Bank by Bank Scaling

DavidFair, 2020-02-17:
**Original reporter:** ISIS/Steve

*Pending input from Steve on how `a b c d` map to banks*

### Expected behavior
Individual bank scaling factors for SANS instruments (mainly LOQ) should be used on a bank-by-bank scaling [docs](https://docs.mantidproject.org/nightly/interfaces/isis_sans/User%20File%20Format.html#id27)

### Actual behavior
The individual scale factors between the HAB and LAB banks are ignored, instead the overall scaling factor. This has been the case going back to the original GUI

smk78, 2020-02-17:
Potentially only affects LOQ & SANS2D as these are the only instruments with more than 1 detector.

smk78, 2020-02-17:
Some industrial archaeology follows...

In COLETTE_SET:
```
c==== ============================================================
c==== 16/11/97 RKH 
c====                           SET SCALES 1.0 2.0 3.0 4.0 5.0
c====  for main, HAB1 to HAB4
c====
      subroutine set_scales

      character*80 text
      character*1 t1,t2
              
      include 'colette_sources:COMCON.INC'
      include 'COLETTE_COMLOQ:'
      include 'COLETTE_SOURCES:COMSMG.INC'
                    
c===============================================================================
c     Output old values
      write (text(1:6), '(f6.3)') scaldet(1)
      write (text(7:12), '(f6.3)') scaldet(2)
      write (text(13:18), '(f6.3)') scaldet(3)
      write (text(19:24), '(f6.3)') scaldet(4)
      write (text(25:30), '(f6.3)') scaldet(5)
      call remark ('OLD SCALES main'//text(1:6)//' HAB1='//text(7:12)//
     > ' H2='//text(13:18)//' H3= '//text(19:24)//' H4='//text(25:30) )

      if (cli$present ('VALUE')) then
         status = cli$get_value ('VALUE', text, len)
      else
         call prompt ('main det scale (zero to keep)==> ', text, len)
      end if
c
      if (len .gt. 0) then
         read (text(1:len), fmt = '(f20.0)', err = 100) value
         if(abs(value).lt.1.e-06)value=scaldet(1)
      else
c==== press return keeps old value
         value=scaldet(1)
      end if
      scaldet(1)=value
c====
      do i=2,5
      write(t1,'(i1)')I
      write(t2,'(i1)')I-1

      if (cli$present ('VALUE_'//t1)) then
         status = cli$get_value ('VALUE_'//t1, text, len)
      else       
         call prompt ('HAB'//t2//
     >       ' relative scale (zero to keep) ==> ', text, len)
      end if
      if (len .gt. 0) then
         read (text(1:len), fmt = '(f20.0)', err = 100) value
         if(abs(value).lt.1.e-06)value=scaldet(i)
      else
c==== zero keeps old value,"prompt" won't get zero length string
         value = scaldet(i)
      end if
      scaldet(i)=value
      end do
c===============================================================
c     Output new values
      write (text(1:6), '(f6.3)') scaldet(1)
      write (text(7:12), '(f6.3)') scaldet(2)
      write (text(13:18), '(f6.3)') scaldet(3)
      write (text(19:24), '(f6.3)') scaldet(4)
      write (text(25:30), '(f6.3)') scaldet(5)
      call remark ('NEW SCALES'//text(1:6)//' rel HAB1='//text(7:12)//
     > ' H2='//text(13:18)//' H3= '//text(19:24)//' H4='//text(25:30) )

c===============================================================================
c     Recalculate spectrum parameters with new lengths  ( altered 25/4/91 RKH)
c==== this done in calling routine if no /NOTABLES
c      i_read=0
c      call set_spec_table(assigned_run,i_read)
c      call set_ring_table                                  

      goto 200

100   call error ('ERROR : '//text(1:len)//' is invalid')
200   call replace_prompt
      return    
      end             
```

So the key variable here is **scaldet(i)**

smk78, 2020-02-17:
In COLETTE_DEC:
```
c=================================================================================
c==== initialise radius (= coordinate) corrections variables
c==== NOTE array size NRC(i) is passed through calls so needs to
c==== start at its MAX possible value           RKH 11/12/92
      x_norm=0
      y_norm=0
      nrc(1)=max_ring
      nrc(2)=max_ring
c
c==== initialise parmeters of SET BANK for HAB detectors  5/9/96
      do i=1,5
      Ihab(i)=303+i
      RLhab(i)=0.0
      ROThab(i)=0.0
      Xhab(i)=0.0
      Yhab(i)=0.0
      end do
c==== initialise the HAB test module icode=308, so that older runs still work
      RLhab(5) = 11.554
      ROThab(5) = 0.0
      Xhab(5) = -112.
      Yhab(5) = -12.
c==== set default effic at 1 Ang for detector path length in COLETTE_CORRECT
      EFFhab = 0.20
      EFFXhab = alog(1.0 - EFFhab)
c==== switch on sample transmission path length correction
      IPATHhab = 1
c==== 16/11/97 set defaults for SET SCALES
      do i=1,5
      scaldet(i)=1.0
      end do
```

smk78, 2020-02-17:
In COLETTE_MONITOR:
```
c===============================================================================
c     Input detctor normalisation file name if given

      ELSE if (cli$present ('FLAT')) then
c==== reset logical name for cell by cell, note this could be set by
C==== default from LOGIN.COM, file is only read in when first needed to be used
C==== by CORRECT or DISPLAY etc., 4/10/95 assume that we want to use it also !
         status = cli$get_value ('FLAT', text, len)
         if(status)then
         status = lib$set_logical( 'INST_FLAT_CELL',text(1:len),,,)
         call remark(' INST_FLAT_CELL file name set to '//text(1:len))
         cell_norm = -1
         end if

c==== cell_norm values are:  0 = not read, 
c====                       -1 = in use but not read yet,
c====                        1 = not in use, 
c====                        2 = by spectrum,  (3 = by detector)
C=== turn off cell by cell normalisation  MON/FLAT/CLEAR
         if (cli$present ('CLEAR').or.cli$present('OFF')) then
           if(cell_norm.ge.2)then
             cell_norm_last = cell_norm
             cell_norm = 1
           else if(cell_norm.eq.-1)then
             cell_norm=0
           end if
         call remark(' INST_FLAT_CELL file will not be used')
c==== 9/12/97 using eff_cell() for scaldet(), need redo tables !
            if(.not.cli$present('NOTABLES'))then
            call set_spec_table(i_sam,0)
            call set_ring_table
            end if
C==== MON/FLAT/ON
         else if (cli$present('ON')) then
           if(cell_norm.eq.0)then
             call remark(' cell by cell normalisation on ')
             cell_norm = -1
c==== reinstate previously cleared off cell by cell normalisation
           else if(cell_norm.eq.1)then
             cell_norm = cell_norm_last
             call remark(' cell by cell normalisation on again')
c==== 9/12/97 using eff_cell() for scaldet(), need redo tables !
            if(.not.cli$present('NOTABLES'))then
            call set_spec_table(i_sam,0)
            call set_ring_table
            end if
           else
c==== cell_norm = -1,2 or 3
             call remark(' cell by cell normalisation already in use')
           end if

         end if
         return
      end if
```
Note the comments here about **eff_cell()** being used for **scaldet()**

smk78, 2020-02-17:
In COLETTE_MAP:
```
c==== 16/11/97 new scalars SET SCALES scaldet(1 to 5) added
C==== 21/1/98  REVISE THESE SO THAT the HAB is also multiplied by scaldet(1)
c
      if(abs(aa(1)).lt.1.e-30.or.abs(bb(1)).lt.1.e-30)then
        call warning('WARNING - sum of raw_eff in SET_SPEC_TABLE is zero ??')
        bb(1)=1.0/scaldet(1)
      else
        bb(1)=bb(1)/(aa(1)*scaldet(1))
      end if
c
      do i=2,5
      if(abs(aa(i)).lt.1.e-30)then
c==== XXXXX put back if use real flat cell effics ever 
c====      call warning('WARNING - summed HAB raw_eff zero ? in SET_SPEC_TABLE')
        bb(i)=1.0/(scaldet(1)*scaldet(i))
c==== 16/11/97 NOTE the raw_eff = -0.9 or -1.1 in flat_cell will still work
c==== in addition to any rescale !
      else
        bb(i)=1.0/(scaldet(1)*scaldet(i)*aa(i))
      end if
      end do

c==== normalise so that solid angle weighted average of EFF_CELL is 1.0
c==== for both ORDELA and HAB separately
        do i_s=1,nsp_total
         if (code_sp(i_s) .eq. 320)then
           eff_cell(i_s)=eff_cell(i_s)*bb(1)
           e_eff_cell(i_s)=e_eff_cell(i_s)*bb(1)
         else if (code_sp(i_s).ge.304 .and. code_sp(i_s).le.308)then
           i=code_sp(i_s)-302
           eff_cell(i_s)=eff_cell(i_s)*bb(i)
           e_eff_cell(i_s)=e_eff_cell(i_s)*bb(i)
         end if
        end do
c
c==== 9/12/97 this is the new part for MON/FL/OFF
c
      else
      do i=1,5
      if(abs(scaldet(i)).lt.1.e-30)then
      call error('ERROR - SET SCALES has a zero ? in SET_SPEC_TABLE')
        bb(i)=1.0
      else
        bb(i)=1.0/scaldet(i)
        if(i.gt.1)bb(i)=bb(i)/scaldet(1)
      end if
      end do
        do i_s=1,nsp_total
         if (code_sp(i_s) .eq. 320)then
           eff_cell(i_s)=bb(1)
           e_eff_cell(i_s)=0.0
         else if (code_sp(i_s).ge.304 .and. code_sp(i_s).le.307)then
           eff_cell(i_s)=bb(code_sp(i_s)-302)
           e_eff_cell(i_s)=0.0
         else
           eff_cell(i_s)=1.0
           e_eff_cell(i_s)=0.0
         end if
        end do
      end if
c
      return                    
      end
```
and
```
c===============================================================================
c     Ring parameters are averages of spectrum parameters
c     weighted by their areas  CHANGED 27/10/94 to use SOLID ANGLES

      do i_r = 1,n_r
         len_r(i_r) = 0.0
         theta_r(i_r) = 0.0
         eff_r(i_r) = 0.0
         e_eff_r(i_r) = 0.0
      end do

      j=0

      do i_r = 1,n_r
         if (rl(i_r+1) .gt. rl(i_r)) then
            do i = rl(i_r),rl(i_r+1)-1  
               i_s = ring_list(i)
c===========   Determine flight path lengths in m.
               len_r(i_r) = len_r(i_r) + len_sp(i_s) * omega_sp(i_s) 
     >                                               / omega_r(i_r)
c===========   Determine scattering angles  
               theta_r(i_r) = theta_r(i_r) + theta_sp(i_s) * omega_sp(i_s)
     >                                                     / omega_r(i_r)
c==== average cell by cell efficiencies if required, RKH 18/11/91
c==== 15/11/94 change to add errors in quadrature, 9/12/97 always do
c==== as eff_cell stores scaldet()
            eff_r(i_r) = eff_r(i_r) + omega_sp(i_s)*eff_cell(i_s)
            e_eff_r(i_r) = e_eff_r(i_r) + omega_sp(i_s)*e_eff_cell(i_s)**2

            end do
         else
            j=j+1
         end if
      end do
```

smk78, 2020-02-17:
In COLETTE_WORK:
```
c===============================================================================
c     COLETTE - LOQ data analysis  program 
c    
c     SHOW_MONITOR also does SHOW_TRANSMISSION      RKH 1/9/87
c
      subroutine show_monitor
      character*16 label(4),lab2(4)
      character*80 text
      integer sys$trnlog

      data label/'SAMPLE      ',
     *             'CAN         ',
     *             'NORMALISER  ',
     *             'BACKGROUND  '/
C
      include 'colette_sources:COMCON.INC'
      include 'colette_sources:COMINS.INC'
      include 'COLETTE_COMLOQ:'
      include 'COLETTE_SOURCES:COMSMG.INC'

c===============================================================================
      status=smg$save_physical_screen(pbid,save_vdid,1)
c      write(6,1000)d_mon,d_det,pre_det
c1000      format(1x,'MON/EFFIC=1 det thick: M2 ',
c     *  F8.3,'mm  main det.',F8.3,'mm',
c     * 1x,'( 1.0 - eff(lam=0)',F8.3)
      aa=sqrt(real(grid_size)+.1)/128
      xx=aa*x_cen/d_dx
      yy=aa*y_cen/d_dy
      write(6,1021)len1_fix,len_det,len_det-len1_fix,len3,x_cen,y_cen,
     *  xx,yy,d_dx,d_dy,scaldet(1)
1021  format(1x,'mod-sam',f7.3,' m,  mod-main det',f7.3,
     *  ' m,  sam-det',f7.3,' m,  mod-mon2',f7.3,' m',/,
     * 1x,'detector centre at x =',f7.2,'mm,  y =',f7.2'mm',/,
     * 1x,'   "        "    col =',f7.2,'   row =',f7.2,/,
     * 1x,'detector pixel size dx =',f8.4,'mm dy =',f8.4'mm',/,
     * 1x,'final SCALE factor is * by ',f8.4,/, 
     * 1x,'( See HELP MONITOR or HELP SET for how to alter these )',/)
c
c==== 5/9/96 add information on HAB high angle bank detectors:
      write(6,651)
651   format(1x,'SET BANK Icode    Distance(m) Rotation(deg)',
     > '    X0(mm)    Y0(mm)  *SCALE')
      do i=1,4
      write(6,652)Ihab(i),RLhab(i),ROThab(i),Xhab(i),Yhab(i),scaldet(i+1)
      end do
      write(6,652)Ihab(5),RLhab(5),ROThab(5),Xhab(5),Yhab(5)
652   format(9x,i6,f12.4,f15.2,2f10.2,f8.4)
      write(6,653)EFFhab
653   format(1x,'effic correction MON/HABEFF= (at 1 Ang ) =',f6.3,
     > ' ( set > 0.99 to ignore)') 
      if(IPATHhab.eq.0)then
      write(6,654)
      else
      write(6,655)
      end if
654   format(1x,'MON/HABPATH/OFF sample transm path correction OFF')
655   format(1x,'MON/HABPATH/ON  sample transm path correction ON')
c
```
and
```
c==== if X units are not Q then write out what we can

*            write(4,'(a,2h W,I3,2x,2a)')text(1:len),n_work,dat_file(n_work),
*     *      run_number(n_work)

            write(4,'(a,2h W,I3)')text(1:len),n_work
[... 90 lines cut ...]
          write(6,2021)
2021  format(1x,'ERROR - cannot continue, too many loops in FIND_HAB ')
          goto 2010
          end if
      x_cen3 = stepx
      y_cen3 = stepy
      x_cen2 = x_cen2+stepx
      y_cen2 = y_cen2+stepy
        status=smg$restore_physical_screen(pbid,save_vdid)
        goto 100
      else if (i.eq.4)then
        x_cen3= 0.0
        y_cen3= 0.0
c==== grab the new scale params for each bank
        do j=1,4
        scaldet(j+1)=scaldet(j+1)*scalb(loop,j)
        end do
c==== don't have to explicitly issue a command, assume mapping tables
c==== reset by final SET BANK command
c      write(command,3005)(scaldet(j),j=1,5)
c3005  format('SET SCALES ',5F8.3)
c
c     status=issue(command)
c
```

smk78, 2020-02-17:
In COLETTE_CORRECT:
```
c===============================================================================
c     COLETTE - LOQ data analysis program - started by R.Osborn  Nov. 1985
c					  - and much modified .....
c
c      this improved normalisation version JAN 1991, RKH
c
c     NOTE error arrays in this routine are almost always variances 
c     i.e. (sigma=standard deviation)**2  unlike GENIE which stores just sigma.
c     So if y is multiplied by CONST then the variance will be multiplied
c     by CONST**2.
c
c     NOTE that GENIE is usually storing histogram data  with y values usually
c     per unit x interval, with the error scaled to match.  It is only in these
c     units that the REBIN_DATA routine will work. 
c
c     Here however we need actual counts in the summations for a particular
c      Q bin so we have to multiply back again by DX after rebinning.
c      
c
c     NOTE our own REBIN_DATA routine assumes the error array to be a scaled 
c          variance.
c
c     Calculate corrected I v Q within prescribed limits
c     Results stored in specified workspace
c
      subroutine isocor(n_work)
C====             THIS ROUTINE FOR ISOTROPIC DATA 
                                                          
.
.
.

C==== NOTE unlike old version, numerator is not now a ratio, we have here
C==== counts per unit wavelength, which have to be converted to counts per
C==== unit Q in both numerator and denominator.  This keeps relative weights
C==== of the data correct from different rings.

C==== I was going to assume sqrt(N) errors on the numerator, but now
C==== note that a separate error is calc above with one BKG option so
C====  should add an extra buffer to the REBIN routine.
C==== For now the e_r( ) above is not used.                 DO THIS -- CHECK

C==== from here on the Error buffers are associated with the denominator 
C====  (unlike the old code)
               dwdq = (x_w(j_w+1) - x_w(j_w) )/( x_b(i_w +1) - x_b(i_w) )
               y_b(i_w) = c_w(j_w,i_sam) * dwdq
               e_b(i_w) = e_trans(j_w,i_sam4)* (w_r(j_w)*dwdq)**2
               w_b(i_w) = trans(j_w,i_sam4)*w_r(j_w)*dwdq

c==== include cell by cell normalisation of detector efficiency RKH 18/11/91
c==== as with other corrections put in denominator so numerator is still
c==== total number of actually recorded counts.
c==== assume e_eff_r is std dev, whilst e_b is variance, the order or
c==== the calc of e_b & w_b is important !  9/12/97 always do, to use scaldet()
c====               if(cell_norm.ge.2)then
       e_b(i_w) = e_b(i_w) * eff_r(i_r)**2 + e_eff_r(i_r)**2*w_b(i_w)**2
               w_b(i_w) = w_b(i_w) * eff_r(i_r)        
c====               end if
c
c==== 2/10/96 larger angle corrections for HAB
             if(i_sam4.eq.i_sam + 4)then
               wav=x_w(j_w)		! xxx should we take mid-bin ??
               scal = 1.0
c==== detector path length increases efficiency
               if(EFFhab.lt.0.99)scal = scal *
     >          (1.0 - exp(EFFXhab*wav*RICOS)) / (1.0 - exp(EFFXhab*wav) )
c==== sample path length lowers transmission
               if(IPATHhab.eq.1.and.alph_w(j_w).gt.-99.0)scal = scal*
[... 52 lines cut ...]
         if (nloops .eq. 4) then
            work(n_work).norm_file = sample(3).file_name
            work(n_work).norm_run = sample(3).run_numb
            work(n_work).back_file = sample(4).file_name
            work(n_work).back_run = sample(4).run_numb
            write (text, '(4(a,i5,a3))')                         
     >      ' SAM:', sample(1).run_numb,ptxt(sample(1).period),
     >      ' CAN:', sample(2).run_numb,ptxt(sample(2).period),
     >      ' NOR:', sample(3).run_numb,ptxt(sample(3).period), 
     >      ' BAC: ', sample(4).run_numb,ptxt(sample(4).period)
         else
            work(n_work).norm_file = ' '
            work(n_work).back_file = ' '
            write (text, '(a,i5,a3,a,i5,a3))')                           
     >      ' SAM: ', sample(i_sam).run_numb,ptxt(sample(i_sam).period),
     >      ' CAN: ', sample(i_sam+1).run_numb,ptxt(sample(i_sam+1).period)
         end if
      else  
         work(n_work).can_file = ' '
         work(n_work).norm_file = ' '
         work(n_work).back_file = ' '
            write (text, '(a,i5,a3))')                           
     >      'RUN: ', sample(i_sam).run_numb,ptxt(sample(i_sam).period)
      end if
```

smk78, 2020-02-17:
I think a discussion with Richard Heenan may be in order.

PR #39806 (2025-09-02): Allow a Separate High Angle Bank Scale Factor to Be Defined in User File
### Description of work

Adds the option to supply a `front_scale` value in the user file. This overrides a value given to `rear_scale` to be applied to the HAB workspaces.

Note: This PR modifies SANS TOML V1. Given that this parameter was already mentioned in the documentation for TOML V1 but didn't do anything, the decision was made to update the old format rather than updating 

Also fixes a test class for the beam centre finder that wasn't in the CMakeLists.txt file and so wasn't being run.

Closes #39341 
Closes #27 [... cut]
commit d68beaf7 (2025-09-02): Allow a Separate High Angle Bank Scale Factor to Be Defined in User File (#39806)

### Description of work

Adds the option to supply a `front_scale` value in the user file. This
overrides a value given to `rear_scale` to be applied to the HAB
workspaces.

Note: This PR modifies SANS TOML V1. Given that this parameter was
already mentioned in the documentation for TOML V1 but didn't do
anything, the decision was made to update the old format rather than
updating

Also fixes a test class for the beam centre finder that wasn't in the
CMakeLists.txt file and so wasn't being run.

Closes #39341 
C [... cut]
closed 2021-09-06 by stale[bot]