mantidproject/mantid #27972: Abins deprecation warning

DanielMurphy22, 2020-02-19:
Found when testing for this issue on the [forum](https://forum.mantidproject.org/t/adding-the-same-lines-to-axes-in-a-different-order-affording-incorrect-plots/648)
### Expected behavior
Abins is used a lot, and we don't want any deprecation warnings
### Actual behavior
Get deprecation warning : `dataset.value has been deprecated. Use dataset[()] instead.`
### Steps to reproduce the behavior
run this:
```
benzene_wrk = Abins(AbInitioProgram="CASTEP", VibrationalOrPhononFile="benzene.phonon",
                    QuantumOrderEventsNumber="1")
for name in benzene_wrk.getNames():
    print(name)
```
you will need this file on your path
[benzene.zip](https://github.com/mantidproject/mantid/files/4224960/benzene.zip)

### Platforms affected
Workbench Nightly since Py3 (Seen on 4.2.20200211.2150)

ajjackson, 2020-02-19:
Good spot, will sort it

ajjackson, 2020-02-20:
Have tracked this down. Only appears with recent versions of h5py, I couldn't see the warnings with the ubuntubionic development docker image until that library was updated with Pip.

These warning messages are pretty unhelpful as they don't give their origin, but it seems that they all come from a couple of places in AbinsModules.IOmodule .

PR #27988 (2020-02-20): Fix deprecation warning: h5py calls from Abins
Closes #27972 

**Description of work.**
See issue #27972 for background.

h5py syntax now requires that values are access by indexing/slicing to encourage efficient access rather than copying large arrays. This is controversial as it now means that `x = dataset.value` should be replaced by more intimidating `x = dataset[()]` even when the value is not an array but, say, a single-character string or a scalar int. This operation is performed many times in Abins as part of the IOmodule routines, leading to a crowded and mystifying error log.

**To t [... cut]
closed 2020-02-20 by gemmaguest