mantidproject/mantid #27994: Slice Viewer not opening on RHEL7 workbench

DavidFair, 2020-02-20:
# Expected behavior
Slice viewer should open on workbench

### Actual behavior
Slice viewer does not open

### Steps to reproduce the behavior
```
from mantid.simpleapi import *
Load(Filename='MAR11060.raw', OutputWorkspace='MAR11060', SpectrumMin=1, SpectrumMax=20)
```
- Attempt to open sliceviewer

### Platforms affected
RHEL7

DavidFair, 2020-02-20:
```
  File "/home/david/mantid/qt/python/mantidqt/widgets/sliceviewer/presenter.py", line 72, in normalization_changed
    self.new_plot()
  File "/home/david/mantid/qt/python/mantidqt/widgets/sliceviewer/presenter.py", line 49, in new_plot_matrix
    self.view.plot_matrix(self.model.get_ws(), normalize=self.normalization)
  File "/home/david/mantid/qt/python/mantidqt/widgets/sliceviewer/view.py", line 130, in plot_matrix
    self.draw_plot()
  File "/home/david/mantid/qt/python/mantidqt/widgets/sliceviewer/view.py", line 135, in draw_plot
    self.colorbar.update_clim()
  File "/home/david/mantid/qt/python/mantidqt/widgets/colorbar/colorbar.py", line 196, in update_clim
    self.cmin_value = data[~data.mask].min()
  File "/usr/lib64/python3.6/site-packages/numpy/ma/core.py", line 3168, in __getitem__
    dout = self.data[indx]
IndexError: in the future, 0-d boolean arrays will be interpreted as a valid boolean index
```

martyngigg, 2020-02-26:
This looks not to be universal and is limited to certain datasets like the one above. I'll keep investigating.

PR #28107 (2020-03-03): Also catch IndexError for numpy array access with bool
**Description of work.**

v1.12 of numpy (on Python 3 on RHEL) raises an IndexError of a scalar boolean is
given to access an array but this was reverted in later
versions. The behaviour should be essentially the same
as there is no mask so we add it to the existing exception
list.

**To test:**

See the original issue and test the file now works and looks the same as a colorfill plot.

Fixes #27994

*This does not require release notes* because **it was broken during the Python 3 move.**

---

#### Reviewer ####

Please comment on the fol [... cut]
closed 2020-03-03 by NickDraper