mantidproject/mantid #28002: Workbench Enum Deprecation warning plotting

DavidFair, 2020-02-21:
Running the following script
```python

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
from mantidqt.plotting.functions import plot

to_plot = []

for i in range(2):
    to_plot.append(CreateSampleWorkspace(OutputWorkspace=str(i), Random=True))

title = 'Tis broken'
ax_properties = {'xscale': 'log',
                 'yscale': 'log'}

plot_kwargs = {"scalex": True,
               "scaley": True}

# Note overplot is true despite it being our first plot
for ws in to_plot:
    plot([ws], wksp_indices=[0], ax_properties=ax_properties, overplot=True,
         plot_kwargs=plot_kwargs, window_title=title)
```

Then clicking on the plot will emit a deprecation warning about a key conversion using the `[]` operator

alicerussell1, 2020-02-21:
I don't get the warning on windows but I do on Ubuntu, have you added the wrong label?

DavidFair, 2020-02-24:
Yes, as it was a Ubuntu VM on Windows :man_facepalming:

closed 2021-03-01 by stale[bot]