mantidproject/mantid #28013: Unscripted Testing workbench

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

StephenSmith25, 2020-02-25:
Help documentation is not loading the stylesheet #28067

DanielMurphy22, 2020-02-25:
We should have more than one person testing Workbench (as we have 4 testing Plot!)
@Harrietbrown found these: #28088 #28082 #28079

StephenSmith25, 2020-02-25:
Extra legend entry appears on Ubuntu when a 1D plot is modified #28090

StephenSmith25, 2020-02-26:
Axes type (e.g) log for a 1D plot can be changed in two ways, which creates undesirable behavior if both are used #28100

StephenSmith25, 2020-02-26:
No more issues found

closed 2020-02-26 by StephenSmith25