mantidproject/mantid #28018: Unscripted Testing Muon

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2020-02-25:
ALC is fine

In muon analysis the sequential fitting fails when doing TF Asymmetry mode https://github.com/mantidproject/mantid/issues/28078

AnthonyLim23, 2020-02-25:
Old muon analysis cannot plot https://github.com/mantidproject/mantid/issues/28080

AnthonyLim23, 2020-02-25:
Muon analysis arrows dont work https://github.com/mantidproject/mantid/issues/28081

AnthonyLim23, 2020-02-25:
Muon analysis does not use the user directories https://github.com/mantidproject/mantid/issues/27593

Table plotting does not work with mixed y and y error https://github.com/mantidproject/mantid/issues/27915

AnthonyLim23, 2020-02-25:
loading PSI data can not be loaded due to workspace wrapper https://github.com/mantidproject/mantid/issues/28089

closed 2020-02-27 by DanielMurphy22