mantidproject/mantid #28020: Unscripted Testing ISIS SANS (old)

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

This is for the old interface. Just check it opens in Mantidplot and runs a very basic reduction. The old instructions (which are too thorough for this simple check are here: http://www.mantidproject.org/SANS_Data_Analysis_at_ISIS

RichardWaiteSTFC, 2020-02-26:
Tested on windows. A few minor issues that may not need to be raised as issues.
(1) I got a warning on opening the interface for the first time (I think this is related to the interface trying to load in the last used user/mask file)
[image]
(2) The 2D reduction gave a workspace containing only zeros for the HAB detector - this may be correct/expected, but it is not the case for the 1D reduction.
(3) Some formatting issues with box placement (see screenshot below)
[image]

DavidFair, 2020-02-26:
Going through
> (1) I got a warning on opening the interface for the first time

Intended by design (yeah I know)

> (2) The 2D reduction gave a workspace containing only zeros for the HAB detector

This is expected, as it looks like there are no counts for the high angle banks

> (3) Some formatting issues with box placement (see screenshot below)

Unfortunately, it's always looked like that hence we have a new shiny clean interface with a nice layout to boot

closed 2020-02-26 by RichardWaiteSTFC