mantidproject/mantid #28028: Unscripted Testing Engineering Diffraction

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: http://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffraction2TestGuide.html

DannyHindson, 2020-02-25:
Just starting testing this UI. The new Python UI isn't currently exposed in Mantid unless you manually edit the mantid.properties file to add Diffraction/Engineering_Diffraction_2.py into the mantidqt.python_interfaces setting. Just discussed with Conor and this sounds like a problem - he's looking into it

DannyHindson, 2020-02-25:
Here are the results of my tests following the scripts. There are a couple of things in here that are suggested changes to the actual test. Also since this screen is in development some items may reflect it's work in progress. I'll discuss with Conor when I'm back in tomorrow afternoon:

Test 1
I had a few comments on the Settings screen:
- could settings screen have a different title to the main window (eg "Engineering Diffraction Analysis **Settings**")?
- the Full Calibration edit box has a red star against it which suggests it's mandatory? But it's only conditionally mandatory
- Help ? icon on settings screen doesn't take you anywhere

Calibrate
- y axis label seems truncated "Fitted Peaks Centre(TOF, us" ? Is there a max char limit?

Focus
- test guide should say "... there is a Focus folder..." rather than "...there is a Calibration folder..."

Test 2 - all fine

Test 3 (Force Vanadium Recalculation)
- I suggest that the test guide should tell the tester to check the datestamp on the files in the Vanadium_Runs folder as well as checking it runs for longer

Test 4 (Cropping)
I got a bit lost on this test:
- What significance does setting the RB Number to "North" have here? Is that  just to avoid overwriting the uncropped results from the earlier tests? Wasn't sure if the RB Number had to match the selected crop
- when I checked "Crop Calibration" and selected Bank 1 (North) I got the pair of plots with title "Engg GUI Vanadium Curves Bank 1" and "... Bank 2" which look the same as the ones from the uncropped calibration. Should I expect to still get bank 2? Also I didn't get the pair of plots with title Engg Gui TOF Peaks". Is that correct? 

Finally one general comment which doesn't relate to any of the 4 tests. Should the "Full Calibration" edit box be populated by any of the steps in the tests? It stayed empty for me

ConorMFinn, 2020-02-26:
### Engineering GUI Changes:
> could settings screen have a different title to the main window

Yes

> a red star against it which suggests it's mandatory

The red star is from the FileFinder widget and indicates an invalid path. Strictly speaking, the input is not mandatory but is so heavily encouraged that having the star there is probably a benefit.

> Help ? icon on settings screen doesn't take you anywhere

I don't see this on my local versions

> y axis label seems truncated

A typo in the plotting functions in the model, easily fixed. 

> Also I didn't get the pair of plots with title Engg Gui TOF Peaks"

This needs fixing before release, it's an issue with the move to quadratic. 

> I got the same results as when selecting Bank 1 (North)

I would expect the results to be very similar.

> setting the RB Number to "North" have here? Is that just to avoid overwriting the uncropped results from the earlier tests?

Yes

> there's an error in the main Mantid event log but nothing in the engineering UI

Some spectra have issues when the peak finding is attempted. This can't really be detected before the algorithm starts. 

### Test guide:
I'll make a quick PR for the changes to the test guide for clarity.

DannyHindson, 2020-02-27:
Retested following Conor's PR and all looks fine now thanks

closed 2020-02-27 by DannyHindson