mantidproject/mantid #28032: Unscripted Testing Sample Transmission calculator

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See http://docs.mantidproject.org/v3.7.2/interfaces/SampleTransmissionCalculator.html

NickDraper, 2020-02-24:
The STC is only on Mantidplot, not workbench (known issue)

It continues to work well, and operate as before
[image]
[image]

NickDraper, 2020-02-24:
tested on windows with the nightly build for mantidplot and master local build for workbench

closed 2020-02-24 by NickDraper