mantidproject/mantid #28033: Unscripted Testing ISIS Reflectometry

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

For testing procedures see http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

DavidFair, 2020-02-25:
Testing on RHEL7:

- Nit pick: Some references to the old GUI in the dev docs for testing
- No warning appears when trying to load a batch file after modifying the table
- Loading the inter batch file, hitting process, then changing the instrument to CRISP then hitting process results in the group rows failing, but the individual rows succeeding?
- Following the above, if you reprocess all runs as INTER the group boxes don't go back to a processed state (still appear as error state)
- In event binning setting the time slices to 30second results in the following error, although this may be me incorrectly using it
```
Property "FIRSTTRANSMISSIONRUN" is not set to a valid value: "Workspace TRANS_13463 is not of the correct type".
Property "SECONDTRANSMISSIONRUN" is not set to a valid value: "Workspace TRANS_13464 is not of the correct type".
```
- Loading a batch file, then trying to exit doesn't give a pop-up asking if the user wants to discard / save their changes
- The ISIS Sample Data INTER file contains a path to a user dir specific to the user who created this in `Save ASCII`

NickDraper, 2020-03-05:
@gemmaguest can you create issues for the things you want to progress from this feedback and close this issue?

gemmaguest, 2020-03-06:
> Testing on RHEL7:
> 
>     * Nit pick: Some references to the old GUI in the dev docs for testing

@DavidFair Could you point them out? I can't see any references to the old GUI other than at the very beginning where it explains the context.

>     * No warning appears when trying to load a batch file after modifying the table

This will be fixed by PR #27655 

>     * Loading the inter batch file, hitting process, then changing the instrument to CRISP then hitting process results in the group rows failing, but the individual rows succeeding?

This will probably be related to the fact that the files are loaded as INTER but it's trying to post-process them as CRISP. This is not something users are likely to do and can be chalked up as user error. The solution is to delete the workspaces so they get re-loaded from the correct instrument (if, indeed, the run numbers are relevant for that instrument). It would probably be better to reset processed state when changing instrument but I don't think it's worth doing.
 
>     * Following the above, if you reprocess all runs as INTER the group boxes don't go back to a processed state (still appear as error state)

Again, this is not intended to be supported. Although it might be just that you need to manually select the group for it to be reprocessed (error rows won't be reprocessed when you process the whole table rather).
 
> * In event binning setting the time slices to 30second results in the following error, although this may be me incorrectly using it
>
>
> ```
> Property "FIRSTTRANSMISSIONRUN" is not set to a valid value: [...]

See issue #23471

>     * Loading a batch file, then trying to exit doesn't give a pop-up asking if the user wants to discard / save their changes

Error in the docs - fixed in PR #28281

>     * The ISIS Sample Data INTER file contains a path to a user dir specific to the user who created this in `Save ASCII`

I've updated the sample data. The directory is now empty.

closed 2020-03-06 by gemmaguest