mantidproject/mantid #28040: Unscripted Testing ISIS SANS (new GUI)

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Particularly try to break the Data Table See: https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html

rprospero, 2020-02-25:
Issue #28063 is major regression that prevents saving reduced data.

rprospero, 2020-02-25:
Issue #28066 is a minor issue that changes behaviour, but does not prevent analyzing data.

rprospero, 2020-02-25:
Closed issue #28063 due to lack of reproducibility.

DavidFair, 2020-02-25:
#28069 fixes #28066

closed 2020-02-27 by rprospero