mantidproject/mantid #28043: Unscripted Testing M-slice

DanielMurphy22, 2020-02-24:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start, this should probably be tested in iDAaaS

Mantid-Matthew, 2020-02-25:
The plotting is m-slice in currently broken have created a high priority issue #28076

AnthonyLim23, 2020-03-03:
@Mantid-Matthew have you completed this testing?

Mantid-Matthew, 2020-03-03:
Plotting now seems to work have found one more issue #28173 but that is all.

closed 2020-03-03 by Mantid-Matthew